TheThingsNetwork/lorawan-stack #1928: Bad payload decryption after a new join when using external JS

Oliv4945, 2020-01-27:
#### Summary
When a device joins with external JS for the 2nd time or more the payload is not decrypted correctly, it seems that session keys used by the LNS are wrong

#### Steps to Reproduce
1. Create a device using[ Semtech's join server](https://www.loracloud.com)
2. Join network
3. Send a payload => In MQTT uplink, payload is ok
4. Reset the device
5. Join the netwok
6. Send a payload => In MQTT uplink, payload is not ok

Reproduced in self hosted v3.5.0 and TTI v3.5.1-rc1

#### What do you want to see instead?
Device payload decrypted correctly each time

#### Can you do this yourself and submit a Pull Request?
No

johanstokking, 2020-01-28:
@Oliv4945 can you confirm that this regression is since v3.5.0; or is v3.4.2 also not working correctly?

Oliv4945, 2020-01-28:
Issue is also in v3.4.2, I tested it yesterday on a clean install

closed 2020-01-28 by johanstokking

Oliv4945, 2020-01-28:
I confirm the fix, thank you😃

PR #1933 (2020-01-28): Generate local session key ID if interop JS didn't

#### Summary

Fix #1928

#### Changes

- According to BI 1.0, either `AppSKey`, or `SessionKeyID`, or both are sent in `JoinAns`
- If `SessionKeyID` is not sent, we still need to have one for AS to detect session changes. We cannot reliably use `DevAddr` for that (as it may return to an old one while the AS wasn't up)
- We are now, in the interop client, generating a `SessionKeyID`. This is prepended with an identifier
- AS uses this identifier to not try to get the `AppSKey` from the interop JS

#### Checklist

- [x] Scope: The referenced is [... cut]