TheThingsNetwork/lorawan-stack #1945: Documentation - Configuartion: ENV variables are defined with columns

LasaleFamine, 2020-02-03:
#### Summary
Configuration doc is reporting the variables within `.env` file with columns but the stack won't start with that env file.

#### Why do we need this ?
It's a bug of the doc or otherwise of the stack parsing the variables in a wrong way, must be fixed.

#### What is already there? What do you see now?
Reference: https://thethingsstack.io/v3.5.1/guides/getting-started/configuration/
As you can see the variables are defined with columns, eg:
```
TTN_LW_TLS_SOURCE="acme"
TTN_LW_TLS_ACME_DIR="/var/lib/acme"
TTN_LW_TLS_ACME_EMAIL="your@email.com"
TTN_LW_TLS_ACME_HOSTS="thethings.example.com"
TTN_LW_TLS_ACME_DEFAULT_HOST="thethings.example.com"
```

#### What is missing? What do you want to see?
Env variables must not have columns when defined within the example:
```
TTN_LW_TLS_SOURCE=acme
TTN_LW_TLS_ACME_DIR=/var/lib/acme
TTN_LW_TLS_ACME_EMAIL=your@email.com
TTN_LW_TLS_ACME_HOSTS=thethings.example.com
TTN_LW_TLS_ACME_DEFAULT_HOST=thethings.example.com
```

#### How do you propose to document this?
///

#### Can you do this yourself and submit a Pull Request?
Yep, just confirm you want to change the doc.

As side note I'm on Ubuntu 18.

Thanks for the awesome work by the way!

johanstokking, 2020-02-06:
How exactly do you use the env file?

LasaleFamine, 2020-02-06:
As specified here https://thethingsstack.io/v3.5.1/guides/getting-started/configuration/, I'm using the `env_file: '.env'` inside the `docker-compose`.

neoaggelos, 2020-02-20:
With #1980 this is no longer relevant.

closed 2020-02-20 by neoaggelos