TheThingsNetwork/lorawan-stack #1948: Support Downlinks via the Console

KrishnaIyer, 2020-02-05:
#### Summary

Support Downlinks via the Console

#### Why do we need this?

To be on par with V2, also non-technical individuals can't be expected to setup the CLI (or use HTTP/MQTT hooks) only to send Downlinks.

#### What is already there? What do you see now?

Downlinks only via API (HTTP/MQTT) or CLI.

#### What is missing? What do you want to see?

Support Downlink operations via the Console

#### Environment

`ttn-lw-stack v.3.5.1`

#### How do you propose to implement this?

🤷‍♂ 

#### Can you do this yourself and submit a Pull Request?

I can help deploy and test.

bafonins, 2020-02-05:
What are the APIs to be used?
Any propositions on where is the right place for this?
How important is this feature?
Can we use the v2 console UI as a reference?
[image]

cc @johanstokking

adriansmares, 2020-02-05:
Regarding the API used for downlink queue operations - use [`AppAs`](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/api.md#service-appas) - The console already uses it partially in the integrations page for the `GetMQTTConnectionInfo` method.

johanstokking, 2020-02-06:
Let's indeed start with simple push and replace operations, with a single message that you can enter as hex. That feature matches with V2.

We can add listing the downlink queue later. That's a super nice feature too.

willhelmx, 2020-02-10:
Really nice if you Implement this features guy´s
Also for developments purposes it would be very nice to have the possibility to se the Payload/ decoded payload in a terminal window like in the public TTN instead of having to click and look at the entire data in every instance

BR
/A

bafonins, 2020-02-11:
@willhelmx We already consider displaying decoded payload and other useful entity/event fields in the events pane for each event. You can find the discussion in https://github.com/TheThingsNetwork/lorawan-stack/issues/1962, do not hesitate to join and express your needs/opinions.

kschiffer, 2020-04-29:
Closed via #2338

PR #2338 (2020-04-23): Schedule downlinks in the Console
#### Summary
Closes #1948 
References #2230 

[image]

#### Changes
- Add downlink queue service to JS SDK
- Add downlink form component
- Add device messages view
- Add byte utilities module
- Enable byte inputs without max value
- Make tabs hideable

#### Notes for Reviewers
The device needs to be activated for the push/replace functions to work. The easiest way to test this is therefore with ABP activated devices.

Byte input for byte values with unlimited length is not optimal yet. I'll follow up with a refactor that will fix this and add more useful funct [... cut]
closed 2020-04-29 by kschiffer