TheThingsNetwork/lorawan-stack #1954: Store downlink queue per session

johanstokking, 2020-02-05:
#### Summary

Store downlink queue per session

#### Why do we need this?

To avoid hitting AS to recalculate the downlink queue when the end device confirms the session. Having a downlink queue of both the current session and the pending session in NS allows for downlink items to be available immediately.

#### What is already there? What do you see now?

Current flow is that on device activation, the new session is stored as pending session in AS. Newly scheduled downlink items are, however, encrypted with the current session. When the end device sends the first uplink message in the session, AS recalculates the downlink queue, but this is not fast enough to make the RX window.

#### What is missing? What do you want to see?

Store the downlink queue in the end device session. This changes the protos, but it should not change the API; the downlink queue operations towards AS should remain the same.

#### How do you propose to implement this?

Add a downlink queue to `Session`, so it's available in the end device's `session` and `pending_session`. You can use the existing downlink queue in `EndDevice` as the current session's downlink queue.

AS should change behavior; it should schedule downlink items for both the current and the pending session, and NS should store the items in the concerning session.

Whenever a device confirms the session to NS, NS and AS switch the session, but AS does not longer need to recalculate the downlink queue.

#### Can you do this yourself and submit a Pull Request?

Can review. Please coordinate. I would suggest that @rvolosatovs starts on the NS side, and @adriansmares basically removes the recalculation on session confirmation in AS.

johanstokking, 2020-02-10:
Blocking #1971

rvolosatovs, 2020-02-10:
I'd prefer merging #1611 before working on this. There are quite some downlink-related changes in #1611, which are very likely to cause a bunch of merge conflicts.

PR #2163 (2020-05-15): Pending session downlink queue support in the Application Server

#### Summary

Closes #1954 
Closes #1971

#### Changes

- Downlink queue operations are now done on both sessions (pending + current).
- Downlink queue listing returns only the queue of the current session.
- On join-accept, the pending downlink queue is recomputed, but the current downlink queue is unchanged.
- Full downlink queue invalidations (`DownlinkQueueInvalidated` event) and individual downlink message nacks result in the computation of the both queues, if the pending session is present. If there is no pending session,  [... cut]
closed 2020-05-15 by adriansmares