TheThingsNetwork/lorawan-stack #1963: Event error payloads are not shown

johanstokking, 2020-02-07:
#### Summary

Event error payloads are not shown

#### Steps to Reproduce

1. Subscribe to gateway or application events in the Console
2. Observe failure or packet drop events; they are not shown
3. Subscribe to the same events in CLI; they are shown

#### What do you see now?

Console doesn't show error payloads:

[image]

#### What do you want to see instead?

Error payloads, like CLI:

```json
{
  "name": "gs.up.drop",
  "time": "2020-02-07T12:30:40.588851469Z",
  "identifiers": [
    {
      "gateway_ids": {
        "gateway_id": "kerlink-ifemtocell",
        "eui": "7276FF003903007D"
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
    "namespace": "pkg/gatewayserver",
    "name": "host_handle",
    "message_format": "host `{host}` failed to handle message",
    "attributes": {
        "host": "cluster"
      },
    "cause": {
      "namespace": "pkg/networkserver",
      "name": "device_not_found",
      "message_format": "device not found",
      "correlation_id": "df971dc6e7c5402596576816401ade98",
      "code": 5
    },
    "code": 5
  },
  ...
}
```

#### Environment

Console

#### How do you propose to implement this?

- Remove a filter somewhere that suppresses error payloads (?)
- Render the error nicely. Would be nice to show a tree of causes

#### Can you do this yourself and submit a Pull Request?

Can review

pgalic96, 2020-02-10:
@johanstokking Does this entail also non uplink/downlink related event errors (such as failed link, failed pubsub etc.)?

htdvisser, 2020-02-10:
This concerns all events with a `"data"` object that has `"@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails"`

PR #1967 (2020-02-10): Display event error payloads

#### Summary

Closes #1963 

#### Changes

- Add the logic for displaying event data to all event component types
- Add the `ErrorMessage` case for displaying event data

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [ ] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [ ] Documentation: Relevant documentation is [... cut]
closed 2020-02-10 by pgalic96