TheThingsNetwork/lorawan-stack #1969: Loading System Root CAs on Windows

htdvisser, 2020-02-10:
#### Summary

When using a custom CA, The Things Stack doesn't load system root CAs on Windows.

#### What is already there? What do you see now?

The CLI and Component try to load the system root CAs, and add custom certs. If loading system root CAs fails, an empty cert pool is used:

https://github.com/TheThingsNetwork/lorawan-stack/blob/05807ed3bbcf4d4c458b113f37df130c36ee2572/cmd/ttn-lw-cli/commands/root.go#L123-L125

https://github.com/TheThingsNetwork/lorawan-stack/blob/05807ed3bbcf4d4c458b113f37df130c36ee2572/pkg/component/tls.go#L171-L173

Since it always fails on Windows (https://github.com/golang/go/issues/16736), The Things Stack may refuse to send webhooks if a custom CA is configured (for example the Packet Broker CA).

#### What is missing? What do you want to see?

We either need to work around this by embedding root certificates (like we did in v2 with https://github.com/TheThingsNetwork/go-utils/tree/master/roots) or document that Windows deployments need to manually add global roots to the configured CA file.

#### Environment

GOOS=windows

johanstokking, 2020-02-10:
We can also discontinue TTS support for Windows

htdvisser, 2022-07-28:
This can be closed. As of Go 1.18 the SystemCertPool is available on Windows.

closed 2022-07-28 by htdvisser