TheThingsNetwork/lorawan-stack #1971: Application Server fails to recalculate downlink queue on joins

rvolosatovs, 2020-02-10:
#### Summary

Application Server fails to recalculate downlink queue on joins.
Refs #1954 

#### Steps to Reproduce

1. Enqueue a bunch of downlinks to a device, such that queue accumulates
2. Make the device join
3. Observe queue being dropped

#### What do you see now?

```
DEBUG Recalculating downlink queue to restore downlink queue on join application_uid=test-app count=16 dev_eui=DEADBEEF01020304 device_uid=test-app.test-dev-b join_eui=01020304DEADBEEF namespace=applicationserver network_server=cobalt session_key_id=01702F766D45A98E561A3B319E7AF104
 DEBUG Recalculate downlink queue               application_uid=test-app count=16 device_uid=test-app.test-dev-b namespace=applicationserver network_server=cobalt new_session_key_id=01702F6D0BE16D224EF716753BA21D70 next_a_f_cnt_down=1
  WARN Failed to replace application downlink queue device_uid=test-app.test-dev-b error=error:pkg/networkserver:f_cnt_too_low (FCnt is too low) grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=networkserver request_id=01E0QQCVGFGZNCWJ96JH7P2B03
  WARN Recalculate downlink queue failed; clearing the downlink queue application_uid=test-app count=16 device_uid=test-app.test-dev-b error=rpc error: code = InvalidArgument desc = error:pkg/networkserver:f_cnt_too_low (FCnt is too low) namespace=applicationserver network_server=cobalt new_session_key_id=01702F6D0BE16D224EF716753BA21D70 next_a_f_cnt_down=1
 DEBUG Replaced application downlink queue      device_uid=test-app.test-dev-b grpc_method=DownlinkQueueReplace grpc_service=ttn.lorawan.v3.AsNs namespace=networkserver queue_length=0 request_id=01E0QQCVH9AH0PFW1TA2QECYGV
  WARN Failed to recalculate downlink queue, items lost application_uid=test-app count=16 dev_eui=DEADBEEF01020304 device_uid=test-app.test-dev-b error=rpc error: code = InvalidArgument desc = error:pkg/networkserver:f_cnt_too_low (FCnt is too low) join_eui=01020304DEADBEEF namespace=applicationserver network_server=cobalt session_key_id=01702F766D45A98E561A3B319E7AF104
```

That's the join-accept:
```json
{
  "name": "as.up.join.forward",
  "time": "2020-02-10T14:17:35.553635062Z",
  "identifiers": [
    {
      "device_ids": {
        "device_id": "test-dev-b",
        "application_ids": {
          "application_id": "test-app"
        },
        "dev_eui": "DEADBEEF01020304",
        "join_eui": "01020304DEADBEEF",
        "dev_addr": "00B717F9"
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.ApplicationUp",
    "end_device_ids": {
      "device_id": "test-dev-b",
      "application_ids": {
        "application_id": "test-app"
      },
      "dev_eui": "DEADBEEF01020304",
      "join_eui": "01020304DEADBEEF",
      "dev_addr": "00B717F9"
    },
    "correlation_ids": [
      "as:up:01E0QQCVGF2CEWJV0RMKPVN3D8",
      "gs:conn:01E0QQB2AZD5F88R8N2HDZWGGN",
      "gs:uplink:01E0QQCVA1PGX54T3STWKMXVRA",
      "ns:uplink:01E0QQCVA2Q17MRFE4VRJ9HRSQ",
      "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01E0QQCVA1K321HGYGD0B9R9QT"
    ],
    "received_at": "2020-02-10T14:17:35.503187641Z",
    "join_accept": {
      "session_key_id": "AXAvdm1FqY5WGjsxnnrxBA==",
      "received_at": "2020-02-10T14:17:35.304388899Z"
    }
  },
  "correlation_ids": [
    "as:up:01E0QQCVGF2CEWJV0RMKPVN3D8",
    "gs:conn:01E0QQB2AZD5F88R8N2HDZWGGN",
    "gs:uplink:01E0QQCVA1PGX54T3STWKMXVRA",
    "ns:uplink:01E0QQCVA2Q17MRFE4VRJ9HRSQ",
    "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01E0QQCVA1K321HGYGD0B9R9QT"
  ],
  "origin": "cobalt",
  "visibility": {
    "rights": [
      "RIGHT_APPLICATION_TRAFFIC_READ"
    ]
  }
}
```

#### What do you want to see instead?

Queue successfully recalculated

#### Environment

3.5.2

#### How do you propose to implement this?

It seems that AS uses wrong (old) session key ID when recalculating the queue after join and so NS rejects the queue replacement.

#### Can you do this yourself and submit a Pull Request?

@johanstokking

johanstokking, 2020-02-10:
@rvolosatovs can you provide reproduction steps with `ttn-lw-cli simulate` ?

johanstokking, 2020-02-10:
I can reproduce as follows;

---

Create a simulation gateway:

```bash
$ ttn-lw-cli gtw create simulate --user-id admin --frequency-plan-id EU_863_870
```

Create new device in linked application:

```bash
$ ttn-lw-cli dev create app1 dev1 --frequency-plan-id EU_863_870 --lorawan-version 1.0.2 --lorawan-phy-version 1.0.2-b --join-eui 1122334455667788 --dev-eui 1122334455667788 --with-root-keys --resets-join-nonces
```

Set `APP_KEY` to the returned `app_key`

Activate the device:

```bash
$ ttn-lw-cli simulate join-request --join-eui 1122334455667788 --dev-eui 1122334455667788 --app-key $APP_KEY --dev-nonce 0000 --lorawan-version 1.0.2 --lorawan-phy-version 1.0.2-b --frequency 868100000 --spreading-factor 7 --bandwidth 125000 --gateway-id simulate
```

Get the session:

```bash
$ ttn-lw-cli dev get app1 dev1 --session --pending-session
```

And set `DEV_ADDR`, `NWK_S_KEY` and `APP_S_KEY` to the given values

Schedule downlink:

```bash
$ ttn-lw-cli dev downlink push app1 dev1 --frm-payload 0101 --f-port 1
```

Confirm the session:

```bash
$ ttn-lw-cli simulate uplink --dev-addr $DEV_ADDR --f-nwk-s-int-key $NWK_S_KEY --app-s-key $APP_S_KEY --f-cnt 2 --f-port 1 --frm-payload 0101 --lorawan-version 1.0.2 --lorawan-phy-version 1.0.2-b --frequency 868500000 --bandwidth 125000 --spreading-factor 7 --band-id EU_863_870 --gateway-id simulate
```

All good, the uplink is received.

However, try to schedule downlink again:

```
$ ttn-lw-cli dev downlink push app1 dev1 --frm-payload 0101 --f-port 1
```

This fails with:

```
 DEBUG Using access token (valid until 5:17PM)
 DEBUG ccResolverWrapper: sending update to cc: {[{localhost:8884  <nil> 0 <nil>}] <nil> <nil>}
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc00065f850, {CONNECTING <nil>}
 DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc00065f850, {READY <nil>}
 DEBUG Finished unary call                      duration=76.197772ms error=rpc error: code = InvalidArgument desc = error:pkg/networkserver:f_cnt_too_low (FCnt `1` is lower than the minimum `2`) error_correlation_id=cbe9a600bc1a4a2c90942e6fee522b36 error_name=f_cnt_too_low error_namespace=pkg/networkserver grpc_code=InvalidArgument grpc_method=DownlinkQueuePush grpc_service=ttn.lorawan.v3.AppAs namespace=grpc
error:pkg/networkserver:f_cnt_too_low (FCnt `1` is lower than the minimum `2`)
    min_f_cnt=2
    f_cnt=1
    correlation_id=cbe9a600bc1a4a2c90942e6fee522b36
```

---

What happens in logs is that NS schedules `DevStatusReq`, at which point NS should inform that the downlink queue is invalidated. But the queue is empty; it is already being recalculated by AS at that moment. It should be recalculated again, but that doesn't happen.

I think this will be fixed automatically with #1954

johanstokking, 2020-02-10:
Blocked by #1954 for now; let's re-evaluate when that's closed

johanstokking, 2020-02-10:
Useful diff;

```diff
diff --git a/pkg/networkserver/errors.go b/pkg/networkserver/errors.go
index dd0ae18e3..b1d897cd3 100644
--- a/pkg/networkserver/errors.go
+++ b/pkg/networkserver/errors.go
@@ -34,7 +34,7 @@ var (
 	errEncodePayload              = errors.Define("encode_payload", "failed to encode payload")
 	errEncryptMAC                 = errors.DefineInternal("encrypt_mac", "failed to encrypt MAC commands")
 	errExpiredDownlink            = errors.DefineFailedPrecondition("downlink_expired", "queued downlink is expired")
-	errFCntTooLow                 = errors.DefineInvalidArgument("f_cnt_too_low", "FCnt is too low")
+	errFCntTooLow                 = errors.DefineInvalidArgument("f_cnt_too_low", "FCnt `{f_cnt}` is lower than the minimum `{min_f_cnt}`")
 	errInvalidAbsoluteTime        = errors.DefineInvalidArgument("absolute_time", "invalid absolute time set in application downlink")
 	errInvalidChannelIndex        = errors.DefineInvalidArgument("channel_index", "invalid channel index")
 	errInvalidConfiguration       = errors.DefineInvalidArgument("configuration", "invalid configuration")
diff --git a/pkg/networkserver/grpc_asns.go b/pkg/networkserver/grpc_asns.go
index 6570d6fef..89d367f42 100644
--- a/pkg/networkserver/grpc_asns.go
+++ b/pkg/networkserver/grpc_asns.go
@@ -145,7 +145,7 @@ func validateApplicationDownlinks(session ttnpb.Session, macState *ttnpb.MACStat
 			return unmatchedQueue, unmatchedDowns, errApplicationDownlinkTooLong
 
 		case down.FCnt < minFCnt:
-			return unmatchedQueue, unmatchedDowns, errFCntTooLow
+			return unmatchedQueue, unmatchedDowns, errFCntTooLow.WithAttributes("f_cnt", down.FCnt, "min_f_cnt", minFCnt)
 
 		case !bytes.Equal(down.SessionKeyID, session.SessionKeyID):
 			return unmatchedQueue, unmatchedDowns, errUnknownSession
```

PR #2163 (2020-05-15): Pending session downlink queue support in the Application Server

#### Summary

Closes #1954 
Closes #1971

#### Changes

- Downlink queue operations are now done on both sessions (pending + current).
- Downlink queue listing returns only the queue of the current session.
- On join-accept, the pending downlink queue is recomputed, but the current downlink queue is unchanged.
- Full downlink queue invalidations (`DownlinkQueueInvalidated` event) and individual downlink message nacks result in the computation of the both queues, if the pending session is present. If there is no pending session,  [... cut]
closed 2020-05-15 by adriansmares