TheThingsNetwork/lorawan-stack #1973: Add different styling for error events in the console event log

pgalic96, 2020-02-10:
#### Summary

Currently error actions are displayed visually the same as succeeded actions, which makes it hard to distinguish whether the actions is a fail or success

#### Why do we need this?

Better UI/UX

#### What is already there? What do you see now?

There are events in the event log, that can be distinguished whether they are error or not by `getEventDataType` which checks `@type` in the event payload.

#### What is missing? What do you want to see?

Visual indicator that notifies the error event.

#### How do you propose to implement this?

- Separate errored actions in a separate event (sub)component type, and add a different icon or different color

#### Can you do this yourself and submit a Pull Request?

Yes

pgalic96, 2020-02-25:
@kschiffer We discussed this a bit at the status meeting, and at first the idea is to just highlight the events with light red - if you have some better idea for the visual indicator let me know. Also would be good to display a different icon. Later we would like to display the error events with their own respective fields. I will work on this these days.

kschiffer, 2020-02-26:
Indeed, let's make the text color red (including the icon) and change the icon to something appropriate. We also need to add a text color variable for that.