TheThingsNetwork/lorawan-stack #1988: Multitech Conduit MTCDT-247A - SAMR34 Fail to join the network

Lucianovici, 2020-02-13:
Hi guys,

I'm very impressed by your modular lora stack v3. I appreciate your work.
Please help me out to understand this issue, that could be considered a bug, but not sure if it is in TTN NS or Multitech's standard packet forwarder v4.0.1-r26.0

## Steps to Reproduce

1. Provisioned a new VPS
1. Follow the steps to [install TTN v3 stack](https://thethingsstack.io/v3.5.1/guides/getting-started/)
1. Create a new application and device in TTN web console.
1. Of course make sure the auth keys are right.
1. Upload demo firmware on end-node with SAMR34
1. Perform join request from end-node.

**This setup with the same gateway, same end-node and same firmware is proven to work using OTAA on the community TTN NS, but [different packet forwarder](https://github.com/kersing/multitech-installer)**

## What do you see now?

**From my end-node demo interface**
[image]

**Gateway Data view**
[image]

**Application Data view**
[image]

**docker-composer logs**
https://pastebin.com/XjG8RzSL

**multitech packet forwarder logs**
https://pastebin.com/xPLbRFZz

## What do you want to see instead?
Well it seems like the join response is delayed a bit by 5 seconds, if you look to the above Data view - that was a surprise since default `RX_DELAY1` is 1 second and `RX_DELAY2` is 2 seconds.
I'd like to see my device successfully joining the network.

## Environment

- Ubuntu 16.04 VPS.
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial
```

- Multitech MTCDT-247A, firmware v5.1.2 - with packet forwarder configured as specified in this PR https://github.com/TheThingsNetwork/lorawan-stack/pull/1793 
[image]

- Multitech relevant lora packages

[image]

- Proof that lora package is running

[image]

- Band: EU868
- LoRaWAN 1.0.2 MAC with 1.0.2 Rev B PHY
- TTN Stack v3.5.1

## How do you propose to implement this?
Could be there is a mixed issue between network latency and gateway backhaul or forwarder. The VPS is hosted in Germany and I'm connecting the gateway in Romania (on high speed ethernet internet). Also tested with a 3G router. No idea how to fix it, would like to go deeper.

## Can you do this yourself and submit a Pull Request?
I'd love to help, but first let's see if it's a TTN or Multitech standard packet forwarder issue. Would you please help me to investigate?

## Notes:

- I am aware of [this](https://www.thethingsnetwork.org/forum/t/microchip-samr34-joining-denied/22109/46) thread, but my problem seems different.
- I am using the default setup for [2.1.9 EU863-870 Default Settings](https://lora-alliance.org/sites/default/files/2018-05/lorawan_regional_parameters_v1.0.2_final_1944_1.pdf#page=28)

## Questions:
1. Why do we have a 5 seconds between `receive uplink message` and `transmit downlink message successfully`? Can this be the reason of failing to join?
2. How can I further investigate to make sure that the gateway indeed received the downlink from NS and transmit it to the end-node?

Thank you!

johanstokking, 2020-02-13:
Thanks for the extensive report.

Looks like the join-accept arrives on time and correctly on the gateway:

```
JSON up: {"rxpk":[{"tmst":3510471868,"chan":0,"rfch":0,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":11.8,"rssi":-30,"size":23,"data":"AEybAtB+1bNwEQAAAO++rd7gnq/29nE="}]}
INFO: [up] PUSH_ACK received in 43 ms
INFO: [down] PULL_ACK received in 48 ms

[...]
  
JSON down: {"txpk":{"imme":false,"tmst":3516471868,"freq":869.525,"rfch":0,"powe":27,"modu":"LORA","datr":"SF9BW125","codr":"4/5","ipol":true,"size":33,"ncrc":true,"data":"IHzW/AF8xI57qMMxV2IBC/zh8tkA6JfVL+kvcp2zRmnU"}}
INFO: tx_start_delay=1493 (1493.562500) - (1497, bw_delay=1.500000, notch_delay=1.937500)
INFO: [down] PULL_ACK received in 48 ms
```

The join-accept is 6 seconds after the join-request in RX2. The join-accept delay is always 5 seconds, so RX2 is 6 seconds.

What I notice is that the data rate for the join-accept in RX2 is SF9BW125, while I suppose it should be SF12BW125. What frequency plan are you using exactly on the end device and the gateway?

Also, can you show the output of `$ ttn-lw-cli end-device get [app-id] [dev-id] --mac-settings --mac-state.desired-parameters` ?

Lucianovici, 2020-02-13:
Hi Johan,

Thanks for your fast response. OK, I understand why the 5 second delay for transmitting the downlink. I started from scratch, since it is very hard to believe it is happening :)

So, here is what I did.

### 1. Recreate my device from TTN Web console

[image]

### 2. Updated keys on my demo firmware

[image]

https://pastebin.com/hGWWhK01

### 3. Attempt to join

Again I get the `LORAWAN_RADIO_BUSY` error.

[Gateway forwarder logs](https://pastebin.com/Pp4EyvFx)
[End node mac settings (you requested)](https://pastebin.com/FMpzny7r)

### 4. Turn on my [TTIG](https://www.thethingsnetwork.org/docs/gateways/thethingsindoor/) hooked @ Community NS, nearby the Multitech gateway

It works well, with the same firmware, of course different keys.
[image]

Yeah I notice it is `SF9` - not sure why, maybe it's normal? However I see that the default spreading factor is set to `SF12` in internal SAMR34 radio attributes.

Do you think this can make the difference? Should I try to force it to `SF12` when joining?

Thank you. I appreciate your help.

johanstokking, 2020-02-14:
@rvolosatovs any idea why we're using SF9 in RX2 join-accept?

@Lucianovici which frequency plan are you using on the device and on the gateway exactly? `EU_863_870` or `EU_863_870_TTN`?

Lucianovici, 2020-02-14:
Hmm :) I wanted to ask - what is the difference? I'm using `EU_863_870_TTN` for the gateway.

[image]

And on the device I'm using `ISM_EU868` band, documented [here](http://ww1.microchip.com/downloads/en/DeviceDoc/SAM-R34-R35-Microchip-LoRaWAN-Stack-Software-API-Reference-Manual-DS70005382A.pdf) page 18, Table 2-37. ISM Band Types.

Should I use `EU_863_870` - without TTN?

johanstokking, 2020-02-14:
Can you enable debug logs on the server? Set `TTN_LW_LOG_LEVEL=debug` in env or `log.level=debug` in yaml/flags

johanstokking, 2020-02-14:
> Should I use `EU_863_870` - without TTN?

Can you try?

Lucianovici, 2020-02-14:
OMG! It worked after I switched to `EU_863_870` - for GW as well as for a new device. Why? :)
What's the difference?

Lucianovici, 2020-02-14:
It is important to mention that I have used the same band for GW and for device (configured from TTN web console). i.e. `EU_863_870_TTN`, now that I changed the GW to `EU_863_870`, and created another device with `EU_863_870` - **it just worked**.

Also, I use LoRaWAN 1.0.2 MAC with 1.0.2 Rev B PHY (for future references)

johanstokking, 2020-02-14:
Great. Then we know what the bug is in our code; join-accepts go in the band's default data rates.

Lucianovici, 2020-02-14:
Alright. Thank you for your help. I'm glad we figure it out :)

closed 2020-02-19 by rvolosatovs