TheThingsNetwork/lorawan-stack #1993: 2D barcode

indusconn, 2020-02-15:
#### Summary

it would be nice to have a 2D barcode representing the Device EUI, Application EUI, App Key
...

#### Why do we need this?

it would be easier to scan and configure the node
...

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### Environment

...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

No
...

johanstokking, 2020-02-17:
Thanks @indusconn. I suppose you mean a QR code.

This is already supported, see https://thethingsstack.io/v3.5.2/guides/generate-qr-code/ and further.

We don't support AppKeys in QR codes, as this is a bad practice.

Although we still need support to generate QR codes in the Console, I'm going to close this issue as this already works as intended.

closed 2020-02-17 by johanstokking

indusconn, 2020-02-18:
Hi @johanstokking 
was thinking more like a QR code in the "device overview" setting page of the application device , you already have a copy shortcut why not make it visual and more easy for setup 
so instead of manually copying and pasting the generated Device EUI, Application EUI, App Key 
then it may be done more easily by scanning 
What I'm trying to do is building an iOS application that can scan the QR from the device setting page and transfer that to the end node via BT or wifi 
Just an Idea.

johanstokking, 2020-02-19:
@indusconn I understand your use case, but you should not be displaying AppKeys in the clear. This is also why show them like passwords, if they're exposed at all. QR codes look scrambled, but it's just plain text, and we will not be supporting that for root keys.

Your iOS application can login via the OAuth flow and use the APIs to get all the information it needs. No need for an intermediate step with QR code scanning.