TheThingsNetwork/lorawan-stack #1994: Frame counter resets too strict

jpmeijers, 2020-02-15:
#### Summary

Some ABP devices save their frame counters in non-volatile storage. When you register this device on TTSv3, the uplinks could be discarded as the frame counter gap is too high.

#### Why do we need this?

Migrating devices to this stack (from TTNv2). Be more relaxed with frame counter. Maybe we don't need this, but I create this issue so that it's a known change from v2 to v3.

In some cases a device could be out of the reach of a gateway for a year, and when it comes back in range its frames could be discarded, even though they are valid.

#### What is already there? What do you see now?

We have a setting that allows frame counter resets. But we still do not allow huge frame counter jumps.

#### What is missing? What do you want to see?

"Ignore frame counter checks" rather than just "Frame counter resets"

#### Environment

The Things Stack Docker image 3.5.3

#### How do you propose to implement this?

Add a second setting that "Ignore frame counter checks" rather than just "Frame counter resets"

#### Can you do this yourself and submit a Pull Request?

Maybe, but not sure if it's a worthwhile feature.

### Debug Logs

```

stack_1      | 2020-02-15T20:27:51.540384589Z  DEBUG Synchronized server absolute time only   gateway_eui=B827EBFFFED88375 gateway_uid=pisupply-shield namespace=gatewayserver/io/udp server_time=2020-02-15 20:27:51.538929305 +0000 UTC m=+1270.886509047 timestamp=3467512755
stack_1      | 2020-02-15T20:27:51.543031039Z  DEBUG Deduplicate uplink                       grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs m_type=UNCONFIRMED_UP major=LORAWAN_R1 namespace=networkserver received_at=2020-02-15 20:27:51.54231537 +0000 UTC request_id=01E158JDVN7GM3NX2WJ57R634W
stack_1      | 2020-02-15T20:27:51.544065570Z  DEBUG Handle uplink                            grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs m_type=UNCONFIRMED_UP major=LORAWAN_R1 namespace=networkserver received_at=2020-02-15 20:27:51.54231537 +0000 UTC request_id=01E158JDVN7GM3NX2WJ57R634W
stack_1      | 2020-02-15T20:27:51.549183872Z  DEBUG Match device                             ack=false adr=false adr_ack_req=false class_b=false dev_addr=26011CE4 f_opts_len=0 f_port=1 frm_payload_len=11 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs m_type=UNCONFIRMED_UP major=LORAWAN_R1 namespace=networkserver received_at=2020-02-15 20:27:51.54231537 +0000 UTC request_id=01E158JDVN7GM3NX2WJ57R634W uplink_f_cnt=62108
stack_1      | 2020-02-15T20:27:51.550729656Z  DEBUG FCnt gap too high, skip                  ack=false adr=false adr_ack_req=false class_b=false dev_addr=26011CE4 device_uid=jpm-crickets.cricket-001 f_cnt_gap=62108 f_cnt_reset=false f_opts_len=0 f_port=1 frm_payload_len=11 full_f_cnt_up=62108 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs last_f_cnt_up=0 m_type=UNCONFIRMED_UP mac_version=1.0.2 major=LORAWAN_R1 max_transmissions=1 namespace=networkserver payload_length=20 pending_session=false received_at=2020-02-15 20:27:51.54231537 +0000 UTC request_id=01E158JDVN7GM3NX2WJ57R634W supports_32_bit_f_cnt=true transmission=1 uplink_f_cnt=62108
stack_1      | 2020-02-15T20:27:51.551479269Z  DEBUG Failed to match device                   ack=false adr=false adr_ack_req=false class_b=false dev_addr=26011CE4 error=error:pkg/networkserver:device_not_found (device not found) f_opts_len=0 f_port=1 frm_payload_len=11 grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs m_type=UNCONFIRMED_UP major=LORAWAN_R1 namespace=networkserver received_at=2020-02-15 20:27:51.54231537 +0000 UTC request_id=01E158JDVN7GM3NX2WJ57R634W uplink_f_cnt=62108
stack_1      | 2020-02-15T20:27:52.744404140Z  DEBUG Done deduplicating uplink                grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs m_type=UNCONFIRMED_UP major=LORAWAN_R1 namespace=networkserver received_at=2020-02-15 20:27:51.54231537 +0000 UTC request_id=01E158JDVN7GM3NX2WJ57R634W
stack_1      | 2020-02-15T20:27:52.744456129Z   INFO Finished unary call                      duration=1.201393188s error=error:pkg/networkserver:device_not_found (device not found) error_correlation_id=54dc35c534554b5e92e13dd5e0a29bf2 error_name=device_not_found error_namespace=pkg/networkserver grpc_code=NotFound grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01E158JDVN7GM3NX2WJ57R634W
stack_1      | 2020-02-15T20:27:52.744468241Z  DEBUG Drop message                             dev_addr=26011CE4 error=error:pkg/gatewayserver:host_handle (host `cluster` failed to handle message) error_cause=error:pkg/networkserver:device_not_found (device not found) gateway_eui=B827EBFFFED88375 gateway_uid=pisupply-shield host=cluster namespace=gatewayserver/io/udp
```

johanstokking, 2020-02-17:
This is how it is specified in LoRaWAN 1.0.3:

[image]

The `MAX_FCNT_GAP` is `16384`.

So regardless of whether I agree this is sane, this is LoRaWAN spec.

This is gone since LoRaWAN 1.0.4 and 1.1.

cc @rvolosatovs

closed 2020-02-17 by johanstokking

rvolosatovs, 2020-02-17:
Indeed, this behavior is dictated by the spec, which is not related to frame counter resets in fact.
In order to allow pre-1.0.4 ABP devices with persistent frame counters, value of which exceed `MAC_FCNT_GAP` to operate on our network, you need to specify appropriate `session.last_f_cnt_up` when creating/updating device.

ElectronicallyE, 2025-05-06:
I've come across this issue as part of my search in order to answer my forum post at https://www.thethingsnetwork.org/forum/t/how-is-frame-count-fcnt-desyncronisation-managed-1-0-4/71298

> The MAX_FCNT_GAP is 16384.
> 
> So regardless of whether I agree this is sane, this is LoRaWAN spec.
> 
> This is gone since LoRaWAN 1.0.4 and 1.1.

What is the limit now in LoRaWAN 1.0.4 and 1.1?

johanstokking, 2025-05-06:
@ElectronicallyE I commented in the issue. In short, for LoRaWAN 1.0.4 and 1.1, the maximum gap is one roll-over. So it depends on the counter values, but the gap is at most 65K. It is not specified, this is common practice in Network Servers (for FCntUp) and end-device stacks (for FCntDown). With older LoRaWAN versions, the 16K limit comes from 16-bit FCnt support, where a lower limit than 65K was required to make replay attacks a bit harder (but still relatively easy).

ElectronicallyE, 2025-05-07:
@johanstokking, thank you for your comments on the TTN Forum. I have asked some additional clarification questions over there to keep communications centralised.