TheThingsNetwork/lorawan-stack #1995: Webhook configured URL trailing slash dropped

jpmeijers, 2020-02-16:
#### Summary

When you configure a Webhook via the console, and the URL ends in a / the / is dropped when the actual call is made.

#### Steps to Reproduce

Webhook is configured as in this example:

[image]

#### What do you see now?

In the docker-compose logs:
```
stack_1      | 2020-02-16T07:33:49.240509848Z  DEBUG Process message                          application_uid=jpm-crickets hook=ttnmapper namespace=applicationserver network_server=ace3b78e8707 url=http://private.ttnmapper.org/integrations/ttn/v3/uplink_message
```

Because the POST is done to an incorrect URL, the server will redirect the call to the correct one that does have a trailing slash. The original call was a POST, but according to the HTTP specs the retry to the correct URL should be a GET. The problem is that the GET does not contain the JSON payload. The server therefore sees an empty request body.

#### What do you want to see instead?

In the logs I want to see:
`url=http://private.ttnmapper.org/integrations/ttn/v3/uplink_message/`

#### Environment

The Things Stack v3.5.3 Docker image

#### How do you propose to implement this?

Webhooks should use the specified URL as is, without stripping the trailing slash.

#### Can you do this yourself and submit a Pull Request?

No

anush0247, 2020-02-16:
+1 we also faced this issue, implemented an internal work around

johanstokking, 2020-02-17:
Thanks for reporting

adriansmares, 2020-02-17:
This is caused by our usage of `path.Join` which is meant for file system paths (which cannot have a trailing slash). 

I'll move this logic (and other usages) to `url.ResolveReference`, which should fix the cause. See https://play.golang.org/p/JhXy_xjAJ6b for an example.

PR #2063 (2020-03-25): Remove inconsistencies in webhook path handling

#### Summary

Closes #1995 

#### Changes

- Changed the way the webhook paths are joined from `path.Join` to `url.ResolveReference`.

#### Notes for Reviewers

First of all, there is an additional bug which currently exist in the latest master which is produced by the current `path.Join` usage: the path URL is being escaped even if it contains a query string. See [this](https://play.golang.org/p/wE-9kZC0oP1) example which can be replicated in master with a base URL of `http://localhost/test` and an uplink message path of `bar?something=else`.

 [... cut]
closed 2020-03-25 by adriansmares