TheThingsNetwork/lorawan-stack #1996: Join Accept webhook data does not contain metadata

jpmeijers, 2020-02-16:
#### Summary

The data received from the Join Accept webhook does not contain metadata like the gateway that received the Join Request and the SF at which the join was attempted.

#### Steps to Reproduce

Set up a Webhook that will send the Join Accept. Perform a Join. Look at the data received from the webhook.

#### What do you see now?

```
{
   "end_device_ids":{
      "device_id":"sodaq-explorer",
      "application_ids":{
         "application_id":"jpm-crickets"
      },
      "dev_eui":"redacted",
      "join_eui":"redacted",
      "dev_addr":"redacted"
   },
   "correlation_ids":[
      "as:up:redacted",
      "gs:conn:redacted",
      "gs:uplink:redacted",
      "ns:uplink:redacted",
      "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:redacted"
   ],
   "received_at":"2020-02-16T07:48:38.272137730Z",
   "join_accept":{
      "session_key_id":"redacted",
      "received_at":"2020-02-16T07:48:38.083605466Z"
   }
}
```

#### What do you want to see instead?

```
{
   "end_device_ids":{
      "device_id":"sodaq-explorer",
      "application_ids":{
         "application_id":"jpm-crickets"
      },
      "dev_eui":"redacted",
      "join_eui":"redacted",
      "dev_addr":"redacted"
   },
   "correlation_ids":[
      "as:up:redacted",
      "gs:conn:redacted",
      "gs:uplink:redacted",
      "ns:uplink:redacted",
      "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:redacted"
   ],
   "received_at":"2020-02-16T07:48:38.272137730Z",
   "join_accept":{
      "session_key_id":"redacted",
      "received_at":"2020-02-16T07:48:38.083605466Z",
      "rx_metadata":[
         {
            "gateway_ids":{
               "gateway_id":"pisupply-shield",
               "eui":"B827EBFFFED88375"
            },
            "timestamp":2663891931,
            "rssi":-65,
            "channel_rssi":-65,
            "snr":8,
            "uplink_token":"redacted",
            "channel_index":3
         }
      ],
      "settings":{
         "data_rate":{
            "lora":{
               "bandwidth":125000,
               "spreading_factor":7
            }
         },
         "data_rate_index":5,
         "coding_rate":"4/5",
         "frequency":"867100000",
         "timestamp":2663891931
      },
   }
}
```

#### Environment

The Things Stack Docker image v3.5.3

#### How do you propose to implement this?

Add extra metadata to join accept model.

#### Can you do this yourself and submit a Pull Request?

No

johanstokking, 2020-02-17:
For the context; what you are seeing is a join-accept, not a join-request. What I suppose is requested here, is to put the metadata from the corresponding join-request in the join-accept.

johanstokking, 2020-08-24:
@adriansmares @rvolosatovs please consider adding `ApplicationJoinRequest` to the upstream messages from NS to AS and up.