TheThingsNetwork/lorawan-stack #2003: Basic Station WS Ping-Pong

KrishnaIyer, 2020-02-17:
#### Summary

Implement WS Ping for Basic Station frontend.

#### Why do we need this?

To prevent our proxy from dropping WS connections due to inactivity of the gateway.
Refs: #1729 

#### What is already there? What do you see now?

No Ping sent.

#### What is missing? What do you want to see?

Send WS pings periodically (configurable).

#### Environment

`ttn-lw-stack v3.5.2`

#### How do you propose to implement this?

Add periodic pongs to BS frontend.

#### Can you do this yourself and submit a Pull Request?

Yes

johanstokking, 2020-02-17:
This is not supported by all Minihub firmwares; afaik not in any publicly released firmware. Enable this only on reported version.

KrishnaIyer, 2020-02-17:
Yes that's true but the server sending a ping (and ignoring pong on versions that don't support it) prevents the proxy from disconnecting the gateways after 60s(now 600s) of inactivity.

johanstokking, 2020-02-17:
OK, as long as we don't expect the pongs!

PR #2013 (2020-02-19): Support WS Ping-Pong for Basic Station

#### Summary

Closes #2003 

#### Changes

- Make Server ping interval configrable
- Add ping-pong handlers
- Tests

#### Notes for Reviewers

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [x] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [ ] Documentation: Relevant documentation is added or updated. [... cut]
closed 2020-02-19 by KrishnaIyer