TheThingsNetwork/lorawan-stack #2006: Support multiple gateway antennas

neoaggelos, 2020-02-17:
#### Summary

Support multiple antennas for a single gateway

#### Why do we need this?

Some gateways may have more than one antennas

#### What is already there? What do you see now?

Support for a single antenna per gateway. Even if multiple antennas exist, only the first one is used and the rest are ignored.

Relevant issues #48, #1820

#### What is missing? What do you want to see?

Handle all gateway antennas.

#### Environment

3.5.3

#### How do you propose to implement this?

Update code in the relevant TODO sections where this issue is referenced

#### Can you do this yourself and submit a Pull Request?

Cannot pick up currently, adding to backlog

johanstokking, 2021-06-23:
This issue is specific to automatically updating the antenna locations based on status messages or uplink metadata. I don't think we really need this.

johanstokking, 2021-06-23:
As this is easy to fix, let's do fix it.

closed 2021-06-25 by johanstokking