TheThingsNetwork/lorawan-stack #2027: Message references are not working between sections

adriansmares, 2020-02-20:
#### Summary

Proto message references in the API reference section of the documentation cannot be clicked if the message is not part of the current section.

#### Steps to Reproduce

1. Go to `Reference` - `Application Webhook APIs`
2. Go to the `ApplicationWebhookIdentifiers` message.
3. Click on the `ApplicationIdentifiers` message type.

#### What do you see now?

The link does not work since the message is not defined in the current document.

#### What do you want to see instead?

The link should lead me to the correct message definition.

#### Environment

1d5aec3

#### How do you propose to implement this?

I'm not sure what would be the best here. cc @htdvisser 

#### Can you do this yourself and submit a Pull Request?

Once we clarify how we can achieve this.

htdvisser, 2020-02-20:
Just add those messages to the page?

adriansmares, 2020-02-20:
If you feel that duplicating them between the sections I'm ok with this :+1:

htdvisser, 2020-02-20:
Yeah we also do that on the other reference pages, for example with the Collaborator message.

johanstokking, 2023-08-05:
@KrishnaIyer is this still relevant?

KrishnaIyer, 2023-08-07:
Nope. I followed the reproduction steps and this is no longer relevant. This is fixed in the docs.

closed 2023-08-07 by KrishnaIyer