TheThingsNetwork/lorawan-stack #2030: Webhook payload codec support

adriansmares, 2020-02-20:
#### Summary

We should support some type of codec for the body of the webhooks, both ingoing and outgoing.

#### Why do we need this?

This allows easier integration with already existing services, which expect a specific format that is incompatible with our `ttnpb.ApplicationpUp` and `ttnpb.DownlinkQueueRequest`

#### What is already there? What do you see now?

The JSON and gRPC formatters that directly marshall to and from `ttnpb.ApplicationUp`.

#### What is missing? What do you want to see?

We should have something like to the device/application payload formatters, but with support for scripting.

#### How do you propose to implement this?

Add support for a script that accompanies `format` and introduce a JavaScript formatter. 

#### Can you do this yourself and submit a Pull Request?

Yes. cc @johanstokking

htdvisser, 2020-02-21:
I'm not sure if this is in scope for The Things Stack.
This can already be solved with Lambda/CloudFunction/NodeRED.
You also don't see other platforms offer this. If Github sends you webhooks, they're in Githubs format. If Slack sends you webhooks, they're in Slacks format. That's just the way it is.

johanstokking, 2020-02-27:
@htdvisser for some necessary context:

1. There's a feature gap w.r.t. integrations between V2 and V3, which only gets bigger:
   - In V2, we allow per-application scripting, so that users can customize the field names and values to how they need it. Sure, the message itself is V2 JSON, but the decoded payload can be customized. On the contrary, in V3, we're pushing the device repository where not the application developer but the device maker decides the payload field structure, names, values and units
   - In V2, we have a bunch of messaging integrations for specific platforms. In V3, we're getting the webhook template repository, which is going to replace that. However, if we don't support any customization of building requests, we have to go to all these platforms to tell them that they should start supporting our format. We can't apply the leverage that GitHub has
2. The plan is to support this in the webhook templates repository only. We will not support per-device or per-application scripts that allow configuration via CLI or Console, or even API for that matter

So in V3, the payload gets decoded by a payload formatter written by device makers (or community), and the request gets built by a script written by the platform developers (or community).

If you want to send data to your own custom application in a public cloud via webhooks, you won't be able to use this feature as that won't be configured via the webhook templates. So those developers will need to adapt our V3 JSON format.