TheThingsNetwork/lorawan-stack #2039: Users w/o RIGHT_APPLICATION_DEVICES_WRITE_KEYS right cannot successfully create/edit end devices

bafonins, 2020-02-24:
#### Summary

The users without the `RIGHT_APPLICATION_DEVICES_WRITE` right cant successfully create/update end devices in the Console

#### Steps to Reproduce

1. Create a new user and add to an application as a collaborator without the `RIGHT_APPLICATION_DEVICES_WRITE` right on that application
2. Login as the new user and try registering a new end device (otaa/abp doesnt really matter here)
3. See the end device not being created in all stack components, form error (similar applies to the edit procedure)

#### What do you see now?

Partial end device creation, no UI is hidden based on user rights

#### What do you want to see instead?

No errors, if the user has the `RIGHT_APPLICATION_DEVICES_WRITE` right it should be possible to create/edit end devices, but not the keys if the `RIGHT_APPLICATION_DEVICES_WRITE` right is missing

#### Environment

v3.5.3

#### How do you propose to implement this?

I see 2 main problems here:

1. UI is not hidden from the user without appropriate rights
2. The js sdk does not rollback failed device creation

#### Can you do this yourself and submit a Pull Request?

yes

bafonins, 2020-03-09:
Regarding creating devices without the `RIGHT_APPLICATION_DEVICES_WRITE_KEYS` right. The problem here is that it is not possible to create an ABP device (more specifically the entry in NS) without session keys, NS will refuse such request; the same applies to multicast devices. On the other hand, it is possible to create OTAA devices without root keys (consider the case when `root_keys` are provisioned on an external join server, while the cluster JS still has the end device stored).

That being said, shall we allow creating devices for users that do not have the right to edit device keys?

We can:
1. Require both `RIGHT_APPLICATION_DEVICES_WRITE_KEYS` and `RIGHT_APPLICATION_DEVICES_WRITE` to create devices via the Console.
2. Imply `RIGHT_APPLICATION_DEVICES_WRITE_KEYS` if the user has `RIGHT_APPLICATION_DEVICES_WRITE`. This, however, breaks the idea that every right is **explicit** in IS.
3. Disable creating ABP/Multicast devices via the Console and allow creating only OTAA devices.

cc @htdvisser @johanstokking

johanstokking, 2020-03-09:
I think we should keep this explicit, so: require `WRITE_KEYS` if you're writing keys only.

The thing is that there are valid scenarios where users create devices without keys, i.e. when they're on an external JS, so we cannot always require `WRITE_KEYS`.

Implicitly setting `WRITE_KEYS` when `WRITE` is set renders `WRITE_KEYS` obsolete. It is designed to make a distinction.

But you're right, for ABP, `WRITE_KEYS` is effectively always required.

bafonins, 2020-03-09:
So, shall we disable creating ABP and multicast devices via the console if the user has no right to write keys?

johanstokking, 2020-03-09:
> So, should we disable creating ABP and multicast devices via the console if the user has no right to write keys?

Yes there's no sensible way to create an ABP device without keys.

closed 2020-03-11 by bafonins