TheThingsNetwork/lorawan-stack #2046: Don't allow LNS Credentials in the GCS CUPS UpdateInfo response if using HTTP

KrishnaIyer, 2020-02-25:
#### Summary

Don't allow LNS Credentials in the GCS CUPS UpdateInfo response if using HTTP.

#### Why do we need this?

The update info response blob shouldn't contain API Keys / Client TLS Creds if communication is plain text.

#### What is already there? What do you see now?

If a request is made to `http://<gcs-host>:<port>/update-info`, we still send back credentials.

#### What is missing? What do you want to see?

Reject the request if it's plain text and the gateway has credentials.

#### Environment

`ttn-lw-stack v3.6.0-rc1`

#### How do you propose to implement this?

- [ ] Check the scheme of the incoming request and reject http requests if LNS creds are to be sent back.
- [ ]  Return `426 Upgrade Required` response  (check if BS client supports this).

#### Can you do this yourself and submit a Pull Request?

Yup

cc: @htdvisser @adriansmares

htdvisser, 2020-02-25:
I don't think it makes a lot of sense to refuse these requests. If there's a MITM, they would already receive the request (including the auth) and can just retry the same request over HTTPS, so in the end we don't improve much.

Practically, I think the HTTP endpoints are mostly used for debugging and on deployments that use secure networks (VPN) so I don't think we should disable returning credentials. On production deployments in the cloud, we should just disable the insecure ports completely so that clients can't even set up a TCP connection.

KrishnaIyer, 2020-02-25:
Ok that's a fair point. 👍

closed 2020-02-25 by KrishnaIyer