TheThingsNetwork/lorawan-stack #2047: Device MAC layer configuration in the console

rvolosatovs, 2020-02-25:
#### Summary

Device MAC layer should be configurable from the console.
Replaces https://github.com/TheThingsNetwork/lorawan-stack/issues/1378

#### Why do we need this?

- Class B device creation/updating from the console
- Creation/updating of end-devices using non-default MAC layer settings (e.g. The Things Uno)

#### What is already there? What do you see now?

CLI as the only means of configuring device MAC layer

#### What is missing? What do you want to see?

Support for configuring all `MACSettings` fields.

##### Field priority list

###### High

- All devices
	- [ ] `rx2_data_rate_index`
	- [ ] `rx2_frequency`
	- [ ] `factory_preset_frequencies`

- Class A-specific (aka, all non-multicast devices)
	- [ ] `rx1_delay`
	- [ ] `rx1_data_rate_offset`
	- [ ] `resets_f_cnt`

- Class B-specific
	- [ ] `ping_slot_periodicity`

###### Medium

- Class A-specific (aka, all non-multicast devices)
	- [ ] `max_duty_cycle`
	- [ ] `supports_32_bit_f_cnt`
	- [ ] `use_adr`
	- [ ] `status_time_periodicity`
	- [ ] `status_count_periodicity`

- Class B-specific
	- [ ] `ping_slot_data_rate_index`
	- [ ] `ping_slot_frequency`
	- [ ] `beacon_frequency`

###### Low

- Class A-specific (aka, all non-multicast devices)
	- [ ] `adr_margin`
	- [ ] `desired_rx1_delay`
	- [ ] `desired_rx1_data_rate_offset`
	- [ ] `desired_rx2_data_rate_index`
	- [ ] `desired_rx2_frequency`
	- [ ] `desired_max_duty_cycle`
	- [ ] `desired_adr_ack_limit_exponent`
	- [ ] `desired_adr_ack_delay_exponent`

- Class B-specific
	- Non-multicast
		- [ ] `class_b_timeout`
		- [ ] `desired_ping_slot_data_rate_index`
		- [ ] `desired_ping_slot_frequency`
		- [ ] `desired_beacon_frequency`

- Class C-specific
	- Non-multicast
		- [ ] `class_c_timeout`

> NOTE: Some of these may already be configurable (e.g. FCnt stuff), please update the checkboxes accordingly and check that these settings are places in correct places (i.e. it's not available for multicast devices)

I believe that class B and C-specific settings should be available for all devices, even for devices for which respective `SupportsClass{B,C}` is `false`. This way the users can (temporary) disable/enable class B/C operation whenever required and keep the specific settings.

Class A-specific settings should only be available for non-multicast devices.

#### How do you propose to implement this?

Add all fields from `MACSettings` https://github.com/TheThingsNetwork/lorawan-stack/blob/74c9da9a9e07a31d7103eabcd3440f9c80c24ea1/api/end_device.proto#L190-L284 as fields to Network Layer configuration. These fields should be configurable at all times, i.e. both on creation and update.

#### Can you do this yourself and submit a Pull Request?

@bafonins will handle it

bafonins, 2020-03-30:
I am mostly done with the **high** priority fields for the MAC Settings configuration:

<details>
<summary>Screenshots</summary>
ABP:

[image]

class B:

[image]

OTAA:
[image]

</details>

However, in order to add all of them and allow the users to register, for example, The Things Uno via the Console the `mac_state.factory_preset_frequencies` field is missing. Im not sure how exactly this should be represented in the UI, currently I have these ideas:

1. As multi-select:
[image]

IMO, such field makes selecting frequencies very easy. Additionally, we can present to the user frequencies based on the `frequency_plan_id` for the end device. However, as @rvolosatovs mentioned that inputs can have arbitrary values and do not necessarily depent on the frequency plan of the end device.

Additionally, for this approach it would be great to have an RPC to fetch presets of frequencies per band.

2. As array of number inputs (similar to how we allow adding headers to webhook integrations):
[image]

This approach is more flexible, however takes more time for the user to set the field, required typing frequencies.

3. As multi-select with options to create new labels:

[image]

Basically, a combination of 1 and 2.

@rvolosatovs @johanstokking

johanstokking, 2020-03-30:
I think having the list (2) is clearest, as that shows the order. And the order is important.

The frequencies can indeed be anything, but it would be very helpful to fetch them via an existing frequency plan.

johanstokking, 2020-03-30:
Adding `compat/api` cause we might need an NS rpc for getting the frequency plan.

bafonins, 2020-03-30:
> I think having the list (2) is clearest, as that shows the order. And the order is important.

Both select components (1) and (2) preserve the order as well. 

> The frequencies can indeed be anything,

With (3) arbitrary frequency values can be added as well

rvolosatovs, 2020-03-30:
I think solution (2) is the cleanest as well, while (3) looks nicer for small amount of frequencies, having multiple (e.g. (4) or more) will look cluttered and hard to keep track of.
It would be great if we could have frequency suggestions for (2) in each textbox (from the proposed frequency plan RPC), so kind of what you see in (3), but per-textbox

bafonins, 2020-07-12:
More or less done with the implementation, but want to wait for https://github.com/TheThingsNetwork/lorawan-stack/issues/2605 to be merged before making a PR to add tests for device wizard (including the mac settings handling)

johanstokking, 2020-11-03:
What is the status here?

bafonins, 2020-11-03:
@johanstokking https://github.com/TheThingsNetwork/lorawan-stack/pull/3065 is ready for review. I added all high and some medium priority fields.

bafonins, 2020-12-21:
Changing milestone of this issue no next up. All high priority fields together with some medium priority fields were added in https://github.com/TheThingsNetwork/lorawan-stack/pull/3065. Will come back to this later if any other fields should be added to the console.

closed 2021-11-30 by bafonins