TheThingsNetwork/lorawan-stack #2070: Allow Brand/Model overrides of Gateway parameters

KrishnaIyer, 2020-02-27:
#### Summary

Allow Brand/Model overrides of Gateway parameters

#### Why do we need this?

Some gateways require some parameters (ex: `clksrc`) to be different from the value set in the FrequencyPlans. These values are brand/model specific.

#### What is already there? What do you see now?

Fixed values for fixed frequency plans.

#### What is missing? What do you want to see?

Override of certain parameters based on model/brand.

#### Environment

`ttn-lw-stack 3.6.0`

#### How do you propose to implement this?

Add hooks on the `GatewayBrand` and `GatewayModel` and update the frequency plan object based on these.

#### Can you do this yourself and submit a Pull Request?

Yup

KrishnaIyer, 2020-03-02:
Ref: https://github.com/TheThingsNetwork/lorawan-stack/pull/1793#issuecomment-593306378

benolayinka, 2020-08-11:
@KrishnaIyer can I bump the priority on this? I'm updating our Multitech guides and this would make configuration of some gateways simpler

KrishnaIyer, 2020-08-11:
This is actually blocked on https://github.com/TheThingsIndustries/lorawan-stack/issues/2018

htdvisser, 2020-08-24:
Back to Backlog until unblocked