TheThingsNetwork/lorawan-stack #2078: Class A downlink timing

johanstokking, 2020-02-28:
#### Summary

This issue describes the technical decision we took for class A downlink timing, which has impact on class B and timed class C downlink as well.

TODO: summary

#### Situation

End devices open the RX1 window for class A downlink `RX1Delay` time [1..15] seconds after transmitting an uplink message. This time can be used for the network to make a roundtrip:

```
ED -> Infr (up) -> Net (up) [-> App (up) -> App (down)] -> Net (down) -> Infr (down) -> ED
```

- Infr(astructure) (up/down): `GW <-> GS [<-> PBA <-> PB <-> PBA]`. Packet Broker part is optional if the device is in the cluster where the gateway is connected
- Net(work) (up/down): `NS`, downlink via downlink task
- App(lication) (up/down): `AS <-> Integration` via rpcs, streams and downlink queue operations. This step is optional as not all class A downlink are data messages

The delays between all these steps are unknown.

Since #1611, NS will never make RX1 if `RX1Delay` is 1 second.

With class B downlink and class C downlink with absolute time, only the downstream part can be considered.

#### Scope

All timed downlink: class A downlink in RX1 and RX2, as well as class B downlink and class C downlink with absolute time. 

#### Criteria

1. Easy of migration from V2
2. Full LoRaWAN support
3. Low battery consumption
4. Distribute available time over the unknowns
5. Easy management

#### Alternatives

0. No change

---

1. Timing settings for NS instance, devices and/or downlink paths

This allows for fine-tuning timing per NS instance, per device and/or per downlink path. It allows for configuring NS instances and devices for specific scenarios, and extends APIs for GS and intermediate infrastructure (Packet Broker) to add meta data that help NS to drive timing decisions

---

2. Simple time division over the unknowns with best effort for making RX1

This is my original idea; divide `RX1Delay` over three components: infrastructure, network and application:

```
rx1Delay = infrastructureDelay + networkDelay + applicationDelay
infrastructureDelay = 1s
networkDelay = (rx1Delay - infrastructureDelay) / 2
applicationDelay = networkDelay = (rx1Delay - infrastructureDelay) / 2
```

This results in even division of time to the network-layer and the application-layer with increasing `RX1Delay`

The delays are two-sided for a class A RX1 roundtrip: in class B and timed class C, divide all delays in 2.

`infrastructureDelay` includes any latencies between the gateway and NS, as well as NS deduplication and downlink scheduling. So, in the ideal world where the latency between the gateway and NS is 0 ms, the deduplication window is 150 ms, the downlink scheduling follows 0 ms after the uplink, the downlink message arrives 850 ms early on the gateway.

`networkDelay` is the time NS schedules the downlink task in advance. This gives time for the application-layer to have scheduled a downlink message, before anything gets scheduled. `networkDelay` includes all time NS needs, including downlink task scheduling and deduplication. If `networkDelay` is 0 (i.e. `RX1Delay` is 1), NS yields all time to the infrastructure and schedules the downlink task immediately.

`applicationDelay` is the time the AS (and upstream) has to respond to a data uplink message.

#### Consideration

The consideration is each alternative per criteria:

| Criteria | Alternative 0 | Alternative 1 | Alternative 2 |
--- | --- | --- | ---
Easy of migration from V2 | no pass | pass | pass
Full LoRaWAN support | no pass; always breaks RX1 with RX1Delay of 1 second | pass | pass
Low battery consumption | rely on RX2 is costly and may require retries when the gateway's RX2 is exhausted | pass | pass
Distribute available time over the unknowns | pass | pass | pass, although we may need to make the 50/50% split configurable in the future
Easy management | pass | no pass; hard to maintain, hard to measure delays with non-synced system clocks, hard to document the trade-offs, hard to change the values for high volumes of devices | pass

#### Decision

Alternative 2, which is already pretty much implemented. This issue serves therefore more as documentation.

#### How do you propose to implement this?

It should be pretty close to what we have. I don't know exactly what the current implementation is, but it seems to be taking NS's `ReceivedAt` into account in the downlink task execution:

> @rvolosatovs: NS will not attempt Rx1 after `ReceivedAt+(Rx1Delay-infrastructureDelay)/2`

If `RX1Delay` is 1, RX1 will never be tried. That's not in the spirit of alternative 2.

Since upstream and downstream are decoupled, indeed, downstream makes an independent decision whether it is still worth trying RX1. This should only be based on NS's `ReceivedAt` and the `RX1Delay`.

Since the infrastructure delay is a sunk cost, NS should yield it to the infrastructure, and not try to make assumptions. It doesn't, can't and shouldn't know how `ReceivedAt` relates to the end device's transmission time. Therefore, NS can only safely consider downlink in RX1 "too late" when the RX1 delay has passed since `ReceivedAt`, regardless of the infrastructure, network and application delays described above. Those only apply to downlink task scheduling.

The reason `TxRequest` contains an RX1 _and_ RX2 option, is for the GS – closest to the gateway – to make the final decision. This is already implemented in the GS scheduler. NS shouldn't start bothering about this too because that's infeasible because of the many unknown latencies.

rvolosatovs, 2020-03-02:
For context, this observation started the whole story:
For class B, the NS should select the next ping slot to attempt downlink on. For that NS needs to know(or estimate) what it the earliest downlink transmission time it could make in order not to waste neither computational resources of our servers, nor gateway scheduling capacity. For this to be possible, NS needs to know the minimum time interval from it sending the downlink to GS to transmission time of that downlink.

johanstokking, 2020-03-02:
> @johanstokking Since the delays are two-sided, for scheduling class B and timed class C downlink tasks, consider half or the full infrastructure delay.

I forgot about the fact that we want to use `RX1Delay` to tune class B and timed class C downlink scheduling. So instead of the above, it should be:

Since the delays are two-sided, for scheduling class B and timed class C downlink tasks, consider `(infrastructureDelay + networkDelay)/2`.

Did we converge to a decision?

rvolosatovs, 2020-04-23:
> networkDelay = (rx1Delay - infrastructureDelay) / 2

> `networkDelay` is the time NS schedules the downlink task in advance

Given these 2, `Rx1Delay == 1` will never make it, since NS will only send the uplink to GS at `up.ReceivedAt.Add(rx1Delay)`, which is always going to be too late.
In any case, I will open a PR soon, which does way more optimistic estimation as described in (2), stay tuned.

johanstokking, 2020-04-24:
> Given these 2, `Rx1Delay == 1` will never make it, since NS will only send the uplink to GS at `up.ReceivedAt.Add(rx1Delay)`, which is always going to be too late.

I see. I think the minimum has to be `infrastructureDelay` then.

johanstokking, 2020-04-27:
Let's leave the label here for future reference. It's still a decision, but not a discussion anymore

johanstokking, 2020-04-27:
Also for reference, #2418 changes:

> Avoid scheduling class B/C downlink tasks for MAC-only frame
>
> That means that NS may still send an `FPort==0` downlink with single `DevStatusReq` at RX1/RX2, but that's the only possibility and IMO correct behavior - the device must open the RX1/RX2 after uplink, so we might as well use these

PR #2418 (2020-05-06): Downlink timing

#### Summary

Closes #2078 
Closes #1637 
Closes https://github.com/TheThingsNetwork/lorawan-stack/issues/2422
Closes https://github.com/TheThingsNetwork/lorawan-stack/issues/2244

#### Changes

- Network Server will attempt RX1 for devices with `Rx1Delay` of 1 second, if possible. In terms of implementation, it means that class A downlink tasks are scheduled ASAP after an uplink, if required.
- Network Server will not attempt to schedule MAC-only frames in ping slots or RXC windows.
- Network Server will only attempt to schedule in a ping slot or RXC window after RX2 has pass [... cut]
closed 2020-05-06 by rvolosatovs