TheThingsNetwork/lorawan-stack #2091: Cache Redis configuration (namespace)

neoaggelos, 2020-03-05:
#### Summary

The cache Redis does not have a namespace by default

#### Why do we need this?

The key naming scheme for the Stack dictates that all Redis keys should exist under a specific namespace. 

#### What is already there? What do you see now?

Default Cache Config is null, so `RootNamespace` is also set to null.

Components that use the cache Redis (e.g. GS and IS) do nothing to check that the Redis config is empty.

#### What is missing? What do you want to see?

- DefaultCacheConfig.Service should probably default to "redis" (which is also the only supported service at the moment.
- DefaultCacheConfig.Redis can be empty by default. However, the components that want to use should check whether the config is empty (using `conf.IsZero()`), and fallback to the default Redis configuration if this is the case.

#### Environment

3.6.0-dev

#### How do you propose to implement this?

- For default cache config, we could add `DefaultCacheConfig{Service: "redis"}`

- For cache Redis namespace: (1) Update all components that use the cache Redis to fallback to the default Redis configuration if needed, or (2) Make this into a `NewRedisCache(config)` and update existing components to use that one. This appears to be a bit more future-proof.

#### Can you do this yourself and submit a Pull Request?

yes

johanstokking, 2020-03-05:
@htdvisser apparently this behavior is consistent with IS. Do we not use the prefix for caching?

htdvisser, 2020-03-05:
There is indeed no default config for the cache (or for events).

In `cmd/internal/shared/events.go` we have the following logic to deal with this:

```go
switch config.Events.Backend {
// ...
case "redis":
	if !config.Events.Redis.IsZero() {
		events.SetDefaultPubSub(redis.NewPubSub(config.Events.Redis))
	} else {
		events.SetDefaultPubSub(redis.NewPubSub(config.Redis))
	}
	return nil
	// ...
```

We could do something similar for the cache:

```go
switch config.Cache.Service {
// ...
case "redis":
	if config.Cache.Redis.IsZero() {
		config.Cache.Redis = config.Redis
	}
	// ...
```

johanstokking, 2020-03-05:
@htdvisser this is specifically about the namespace, i.e. keys not starting with `ttn:v3:...`

htdvisser, 2020-03-05:
The `ttn:v3` namespace comes from the default config: https://github.com/TheThingsNetwork/lorawan-stack/blob/master/cmd/internal/shared/config.go#L83

We don't have that default in the other Redis configs

neoaggelos, 2020-03-05:
I think it would make sense to have a default namespace for the cache as well.

johanstokking, 2020-03-05:
I think so too. And since it's cache, it's a non-breaking change.

PR #2104 (2020-03-11): Setup Redis configuration on initialization

#### Summary

Closes #2103, also closes #2091

#### Changes

- Setup Redis configuration for all sub-systems during the stack initialization

#### Notes for Reviewers

...

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [X] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [ ] Documentation: Relevant [... cut]
closed 2020-03-11 by neoaggelos