TheThingsNetwork/lorawan-stack #2092: Implement user session API handlers

kschiffer, 2020-03-05:
#### Summary
The stack is currently missing handlers for the user session service. These handlers are needed to enable session management via the API.

#### Why do we need this?
It's necessary for the user session management of the account app (#1422)

#### What is already there? What do you see now?
- API protobuff definitions
- Management of authorizations

#### What is missing? What do you want to see?
- Handlers for the rpcs of the `UserSessionRegistry` service (`List` and `Delete`)

#### Environment
v3.6.0

#### How do you propose to implement this?
Trivial.

#### Can you do this yourself and submit a Pull Request?
I could do it, though I'd suggest some of the backenders doing it to be faster

htdvisser, 2020-03-05:
I don't think I'll have time for this in March, so either it's later, or someone else needs to take this.

kschiffer, 2020-03-05:
I'll do then.

PR #4353 (2021-07-12): Implement UserSessionRegistry RPCs

#### Summary

This adds implementations for the UserSessionRegistry RPCs in the Identity Server and CLI.

Closes #2092

#### Testing

- Added unit tests
- Manual testing with the CLI

##### Regressions

None that I can think of.

#### Notes for Reviewers

...

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Docu [... cut]
closed 2021-07-12 by htdvisser