TheThingsNetwork/lorawan-stack #2098: client is not authorized to request a token using this method

cyclops1982, 2020-03-08:
#### Summary
Installed it via getting started guide. Getting
"client is not authorized to request a token using this method" when i try to login to the console.

I'd really like some help with this as i'm new to docker and ttn-stack :)

#### Steps to Reproduce
1. Clear cookies
1. Go to console
2. Enter admin / passwd added during create-admin-account step
3. Get error message "client is not authorized to request a token using this method" on URL https://ttn.lnd.prof-x.net:8885/oauth/authorize?client_id=console&redirect_uri=%2Fconsole%2Foauth%2Fcallback&response_type=code&state=kkm1GRX5cvajZOFu

#### Logs / Configs
Config: http://cyclops.nettrends.nl/stuff/lora/ttn-lw-stack.yml
Docker compose file: http://cyclops.nettrends.nl/stuff/lora/docker-compose.yml
Docker log: http://cyclops.nettrends.nl/stuff/lora/docker-output.txt

#### Environment
[23:15:57] root@jubilee:~/dev/ttnstack/config/stack# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.4 LTS
Release:        18.04
Codename:       bionic

ttn.lnd.prof-x.net is an internal domain, can't be accessed from the outside world.
i'm currently running this on 8885/1885 as 80/443 are currently in use.

johanstokking, 2020-03-12:
Did you create the OAuth client? See https://thethingsstack.io/v3.6.0/guides/getting-started/running-the-stack/

htdvisser, 2020-04-14:
Since there has not been any response from the author of this issue in the last month, we assume that they managed to solve the issue on their own. I'll close the issue now, but feel free to reopen if this is still a problem.

closed 2020-04-14 by htdvisser

cyclops1982, 2020-04-14:
Sorry, yes it can be closed!