TheThingsNetwork/lorawan-stack #2102: Inconsistent handling of session/root keys by NS/AS/JS

bafonins, 2020-03-09:
#### Summary

NS/AS/JS handle update keys request inconsistently for users that have only `RIGHT_APPLICATION_DEVICES_WRITE_KEYS`, but no `RIGHT_APPLICATION_DEVICES_READ_KEYS`. Namely, NS and JS returns the updated keys, while AS ommits them in the response

#### Steps to Reproduce

<details>
<summary>Details</summary>

Update keys in JS with the following payload:
```
{
   "end_device":{
      "ids":{
         "device_id":"lw-100-otaa",
         "application_ids":{
            "application_id":"admin-app"
         },
         "join_eui":"0102030405060708",
         "dev_eui":"0000100000000000"
      },
      "root_keys":{
         "app_key":{
            "key":"F535A60D4BC3C522C5E404C052C04FC2"
         }
      }
   },
   "field_mask":{
      "paths":[
         "root_keys.app_key.key",
         "ids.device_id",
         "ids.application_ids.application_id"
      ]
   }
}
```
See response (note the `app_key` included in the response payload): 
```
{
   "ids":{
      "device_id":"lw-100-otaa",
      "application_ids":{
         "application_id":"admin-app"
      },
      "dev_eui":"0000100000000000",
      "join_eui":"0102030405060708"
   },
   "created_at":"2020-03-09T09:25:17.289544Z",
   "updated_at":"2020-03-09T10:23:38.148097Z",
   "root_keys":{
      "app_key":{
         "key":"F535A60D4BC3C522C5E404C052C04FC2"
      }
   }
}
```

Update keys in NS with the following payload:
```
{
   "end_device":{
      "session":{
         "keys":{
            "f_nwk_s_int_key":{
               "key":"9EE8A50FED138DDBFB1F95FAA037E764"
            }
         }
      },
      "ids":{
         "device_id":"lw-100-abp",
         "application_ids":{
            "application_id":"admin-app"
         }
      }
   },
   "field_mask":{
      "paths":[
         "session.keys.f_nwk_s_int_key.key",
         "ids.device_id",
         "ids.application_ids.application_id"
      ]
   }
}
```
See response (note the `f_nwk_s_int_key` included in the response payload as well nullable `dev_addr` in `session` and proper address in `ids`): 
```
{
   "ids":{
      "device_id":"lw-100-abp",
      "application_ids":{
         "application_id":"admin-app"
      },
      "dev_addr":"01DA969F"
   },
   "created_at":"2020-03-09T09:25:19.562364Z",
   "updated_at":"2020-03-09T10:26:17.798399Z",
   "session":{
      "dev_addr":"00000000",
      "keys":{
         "f_nwk_s_int_key":{
            "key":"9EE8A50FED138DDBFB1F95FAA037E764"
         }
      },
      "started_at":"0001-01-01T00:00:00Z"
   }
}
```

Update keys in AS with the following payload:
```
{
   "end_device":{
      "session":{
         "keys":{
            "app_s_key":{
               "key":"A4A3B5D7B16954F208BBE5EA14040479"
            }
         }
      },
      "ids":{
         "device_id":"lw-100-abp",
         "application_ids":{
            "application_id":"admin-app"
         }
      }
   },
   "field_mask":{
      "paths":[
         "session.keys.app_s_key.key",
         "ids.device_id",
         "ids.application_ids.application_id"
      ]
   }
}
```
See response (note empty `app_s_key` in the response payload as well nullable `dev_addr` in `session` and proper address in `ids`): 
```
{
   "ids":{
      "device_id":"lw-100-abp",
      "application_ids":{
         "application_id":"admin-app"
      },
      "dev_addr":"01DA969F"
   },
   "created_at":"2020-03-09T09:25:19.628866Z",
   "updated_at":"2020-03-09T10:28:02.115582Z",
   "session":{
      "dev_addr":"00000000",
      "keys":{
         "app_s_key":{

         }
      },
      "started_at":"0001-01-01T00:00:00Z"
   }
}
```

</details>

#### What do you see now?

Inconsistency between responses.

#### What do you want to see instead?

IMO, if the has no right to read the keys, but only to set the stack shouldn't return those as AS does. Not sure what is expected behavior with `dev_addr`

#### Environment

v3.6.0

#### How do you propose to implement this?

Most probably NS and JS should take into account the read/write keys right when composing a response.

#### Can you do this yourself and submit a Pull Request?

@rvolosatovs 

cc @htdvisser @johanstokking

rvolosatovs, 2020-03-09:
The assumption in current implementation is that `WRITE` right implies `READ` right, which is apparently not true.

rvolosatovs, 2020-03-10:
Taking another look at the issue it looks like the AS simply does not write the keys to the registry, which is why you're getting an empty keys back. `SET` operations always return the fields actually being `SET`.

johanstokking, 2020-03-12:
> Also note that AS response you provide is clearly invalid. `session.dev_addr` must be the same as `ids.dev_addr`.

It's the same response as NS.

closed 2020-03-12 by johanstokking