TheThingsNetwork/lorawan-stack #2119: External Join Server user experience

johanstokking, 2020-03-10:
#### Summary

External Join Server user experience

cc @laurensslats 

#### Why do we need this?

To make improve the user experience of creating and editing devices on a Join Server.

#### What is already there? What do you see now?

There's currently an inconsistency:

1. In a regional cluster, "External Join Server" equals empty `join_server_address`.
2. However, when using TTI Join Sever, the device should be managed there. That only works when `join_server_address` is the local host name. Because of point 1, the address is empty, so this doesn't work

#### What is missing? What do you want to see?

From a user experience perspective, there are three states:

1. Using the cluster Join Server
   - `join_server_address` is the cluster address
2. Using a global Join Server with known address
   - `join_server_address` is a user-specified host
3. Using a global Join Server with unknown address
   - `join_server_address` is empty

For option 2 and 3, we probably want NS to serve a mapping of JoinEUI to Join Server address, so Console can show a drop down and pre-fill the JoinEUI and Join Server address. cc @rvolosatovs this comes from interop, and probably needs an address field in the JS interop config. If this can be done as part of closing this issue in this milestone, I'm good with that, otherwise for 2 we need to stick with manually entering the JS address.

#### Can you do this yourself and submit a Pull Request?

Can review behavior

rvolosatovs, 2020-03-10:
@johanstokking can you clarify what `join_server_address` is for at all?
> For option 2 and 3, we probably want NS to serve a mapping of JoinEUI to Join Server address, so Console can show a drop down and pre-fill the JoinEUI and Join Server address
`JoinEUI`s uniquely identify the Join Server, what is the use case for *prefilling* or *choosing* the `JoinEUI`? That data is already chosen by the device manufacturer and the user should not alter that. Do we want to provide a way for the user to override LoRa Alliance DNS and cluster configuration for joins?

What I understand from
> There's currently an inconsistency:
>
> 1. In a regional cluster, "External Join Server" equals empty `join_server_address`.
> 2. However, when using TTI Join Sever, the device should be managed there. [...]

Is that the issue is specific TTI Join Server and "External Join Server" checkbox is confusing there.
Why not simply remove/disable the checkbox for TTI Join Server? I suppose that could be a console flag or maybe console could default to disabling the checkbox automatically when JS is the only available component(i.e. NS and AS are disabled in the cluster, while JS is not).
I don't completely understand what is the problem we're trying to solve here.

johanstokking, 2020-03-11:
> @johanstokking can you clarify what `join_server_address` is for at all?

This is like `network_server_address` and `application_server_address`.

It is for API clients (CLI and Console) to figure out where the TTS/TTES Join Server is. For example, the Console only allows editing fields if the `{network|application|join}_server_address` is the local address.

> `JoinEUI`s uniquely identify the Join Server, what is the use case for _prefilling_ or _choosing_ the `JoinEUI`? That data is already chosen by the device manufacturer and the user should not alter that.

The use case is this:
- User U buys device from device manufacturer DM
- User U sees that the root keys are already provisioned on TTI Join Server
- User U therefore selects TTI Join Server in the dropdown
- The device gets created with JoinEUI `70B3D57ED0000000` and `join_server_address` of `[<tenant-id>.]join.cloud.thethings.industries`

> Do we want to provide a way for the user to override LoRa Alliance DNS and cluster configuration for joins?

No

> What I understand from this is that the issue is specific to TTI Join Server and "External Join Server" checkbox is confusing there.

No, it is not specific. We can have the following external Join Servers:

1. The Things Industries Join Server (with `join_server_address`)
2. Semtech Join Server (without `join_server_address`)
3. Anything else

> I don't completely understand what is the problem we're trying to solve here.

See **What is already there**, hopefully it's clear with this explanation.

johanstokking, 2020-03-11:
Another issue is unchecking "External JS" in the Console and saving the device. This does not delete the device from the external JS.

johanstokking, 2020-03-11:
The three cases:

> * Using the cluster Join Server

This is where the user does not use any external join server but just the join server in the cluster. Nothing special, just allow setting JoinEUI, DevEUI and root keys.

> * Using the device already on a global join server (i.e. TTI Join Server)

This is where the user first created the device in the Join Server (i.e. TTI Join Server). That Console only has a Join Server enabled. The user should be able to set JoinEUI, DevEUI, root keys, NetID and KEKs.

It does not make sense to allow creating ABP devices here, nor any NS/AS settings (MAC versions, payload formats, etc), nor choosing for "external JS" (because this JS is already "external" because there is no NS and AS enabled).

Then, the user goes to the regional cluster Console (i.e. TTI Cloud Hosted `eu1`), and the device is already there, because the device is already in the same Identity Server. However, there are no NS and AS settings here, and the NS and AS settings are also still empty. The user now doesn't "create" the device, but should set all the NS and AS settings as if it is being created in NS and AS. 

So, the user should easily indicate to use the current address for NS and AS, and set the NS and AS settings. External JS checkbox remains checked, and disabled.

If the user does not specify any AS fields (because they don't set payload formats for example), the device should still be created in AS. This is called "touch" in CLI.

So, the user experiences a "create", while the device is already created on IS, but it should be created on NS and AS and the NS and AS settings should be set in IS.

> * Using a global Join Server with unknown address

This is for example Semtech Join Server. In this case, the user already created or claimed the device in the external Join Server, and that is out of our scope, because that is not our Identity Server.

In this case, the user actually creates the device in `eu1` via **Create device** and enters the JoinEUI and done.

bafonins, 2020-03-11:
Related https://github.com/TheThingsNetwork/lorawan-stack/issues/1567

rvolosatovs, 2020-05-12:
Is there anything still required from me here?

rvolosatovs, 2020-05-26:
No answer in 2 weeks - removing assignment

johanstokking, 2020-06-24:
Replaced by #2777

closed 2020-06-24 by johanstokking