TheThingsNetwork/lorawan-stack #2130: Basic Station Integration: router_config should not include hardware-specific configuration

beitler, 2020-03-11:
#### Summary
In the Basic Station protocol, the LNS sends a configuration structure router_config down to the solver. Upon reception, Station merges the received configuration with the local station.conf file where the server-side router_config has precedence. TTN servers include configuration inside the LNS-side router_config structure which is highly hardware specific:
`clksrc`
`radio_x.rssi_offset`
`radio_x.tx_enable`

This breaks the configuration on platforms, which don't comply with these assumptions.

reissjason, 2020-03-18:
Will the LUT values be sent by default? They are unique to GW hardware.

KrishnaIyer, 2020-03-18:
@reissjason : TxLut values are not sent for Basic Station gateways.

KrishnaIyer, 2020-09-24:
blocked on https://github.com/TheThingsIndustries/lorawan-stack/issues/2018 unless this is critical enough that we want to hardcode these fields.

Oliv4945, 2021-01-05:
Hello @KrishnaIyer,

It seems that TheThingsIndustries/lorawan-stack#2018 is private. Do you have any update on this issue ?

Thanks

KrishnaIyer, 2021-01-11:
This item is in `backlog` so it's not on my list for this milestone.Please upvote if this is important to you. If there's enough demand, we can definitely look at it.

KrishnaIyer, 2021-01-11:
> TTN servers include configuration inside the LNS-side router_config structure which is highly hardware specific:
clksrc
radio_x.rssi_offset
radio_x.tx_enable
> This breaks the configuration on platforms, which don't comply with these assumptions.

@beitler 
I went through the LBS spec;  https://doc.sm.tc/station/gw_v1.5.html and the following is the example configuration file;
```
   "radio_conf": {                  /* Actual channel plan is controlled by the server */
       "lorawan_public": true,      /* is default */
       "clksrc": 1,                 /* radio_1 provides clock to concentrator */
       "device": "spidev",          /* default SPI device is platform specific */
       "radio_0": {
           /* freq/enable provided by LNS - only hardware-specific settings are listed here */
           "type": "SX1257",
           "rssi_offset": -166.0,
           "tx_enable": true,
           "antenna_gain": 0
       },
       "radio_1": {
           "type": "SX1257",
           "rssi_offset": -166.0,
           "tx_enable": false
       }
       /* chan_multiSF_X, chan_Lora_std, chan_FSK provided by LNS */
   }
```

So for LBS, the fields listed in this example should not be sent down by the LNS as they would be present in the local `station.conf`. Is that correct?
- how about if the server could choose these fields based on the particular gateway/model? Or should the server assume that the gateway should have the base `station.conf` embedded in it when it's released?
- Are there other such fields that the LNS should not override?

beitler, 2021-01-11:
The LNS has the power to set all low level radio configuration parameters. However, the LNS should make use of that power only when it is absolutely necessary and it knows exactly what it is doing.
Historically, the LNS would send down the channel plan via the hardware specific `radio_conf` structure. Since 2.0.5 this is not necessary since the channel plan can be described in a hardware agnostic way using the `upchannels` and `DRs` fields of the `router_config` message.
Therefore, the recommendation is that pre-2.0.5 the LNS specifies inside `radio_conf` only those fields that are necessary to describe the channel plan. Since 2.0.5 there should not be any need for the LNS to specify any `radio_conf` parameters except for some low-level tuning.

kristokj, 2021-02-26:
Hi @beitler and @KrishnaIyer,
Could you please comment on the progress for this matter? 

We are working on deploying several MultiTech Conduit AEPs with basic station on TTNv3, but have so far not been able to get it running correctly due to this issue of override coming down from the LNS server. 
See these forum posts for details:
https://www.thethingsnetwork.org/forum/t/any-way-to-edit-the-ttnv3-gateway-global-conf-json/44847
https://www.thethingsnetwork.org/forum/t/unable-to-connect-multitech-conduit-aep-to-ttn-v3/44257
https://www.thethingsnetwork.org/forum/t/multitech-conduit-ap-v3-basic-station/39612

PR #4190 (2021-05-20): Remove hardware specific fields in LBS Router Config

#### Summary

Remove hardware specific fields in LBS Router Config

- Closes #2130 
- Refs; 
  - https://www.thethingsnetwork.org/forum/t/any-way-to-edit-the-ttnv3-gateway-global-conf-json/44847 
  - https://www.thethingsnetwork.org/forum/t/unable-to-connect-multitech-conduit-aep-to-ttn-v3/44257
  - https://www.thethingsnetwork.org/forum/t/multitech-conduit-ap-v3-basic-station/39612
- Potentially also addresses #1303 (needs testing)

#### Changes

- Remove hardware specific fields
- Testing

#### Testing

UT and gateway testing

**Before**  [... cut]
closed 2021-05-20 by KrishnaIyer