TheThingsNetwork/lorawan-stack #2134: Expose entity attributes in the Console

adriansmares, 2020-03-11:
#### Summary

The console should expose the entity attributes as part of the general information pages. They probably should also be editable as part of their respective settings page.

> Entity refers to applications, devices, gateways, users and organizations.

#### Why do we need this?

The attributes themselves can be used to store entity metadata, which is useful for users that want to store additional information with their entities.

#### What is already there? What do you see now?

There is support for retrieving and setting the entity attributes in the CLI. It works very much like headers do in the webhooks integration, since attributes are stored as a similar key value mapping.

#### What is missing? What do you want to see?

Some way of seeing the current attributes of an entity and setting them.

#### How do you propose to implement this?

For an initial version exposing the key value structure (using the same key value component that the webhook headers use) should suffice.

#### Can you do this yourself and submit a Pull Request?

I can review this. @pgalic96 feel free to transfer this to someone else if you are too busy to pick this up.

ajovertoom, 2020-03-12:
This is a long-awaited request by Wilhelmsen so would be great if the issue could be prioritised.

mjamescompton, 2020-03-13:
As discussed with alexander, The priority here should be to add the ability to add and edit attributes to an application. This is so that the device claiming app can show these as the default values for a device while claiming.

Displaying and editing these per device can be done after and we should break up this task to per entity type, prioritised in this order  applications, devices, gateways, users and organizations

mjamescompton, 2020-03-13:
@adriansmares Do you know if we can already request the application attributes using the jsSDK or is that part of this task.

bafonins, 2020-03-13:
@mjamescompton you can already get/set attributes on entities. this is issue about displaying/setting these via the web ui.

pgalic96, 2020-03-13:
Do we want it as a separate page in application/device tabs?

mjamescompton, 2020-03-13:
I would say yes, as I don't think it will be easy to find under general settings @bafonins @kschiffer ??

bafonins, 2020-03-13:
for applications/gateways/organizations/users it could be a part of the general settings page
for devices imo it is better to put the attributes into a separate section, similar to location

kschiffer, 2020-03-19:
> for applications/gateways/organizations/users it could be a part of the general settings page
for devices imo it is better to put the attributes into a separate section, similar to location

I agree. For now, it's sufficient to display the attributes in the general settings views only (and on a dedicated tab, for devices). I need a better idea of what we are going to use the attributes for, to better tell where else and how these could be shown in the console in the future.

> The attributes themselves can be used to store entity metadata, which is useful for users that want to store additional information with their entities.

So this is mostly for developers to be able to work with these values inside of their applications, right? I remember we also had the plan to use attributes to enable searchable tags for our entities. So it seems like the use can be very universal.

mjamescompton, 2020-03-19:
>  I need a better idea of what we are going to use the attributes for

I will be using the application attributes as the device claiming attribute form fields, though this feel like a temporary solution as there are most likely other reasons to use application attributes. I thought I might throw out an idea. I was thinking of pre defined key prefixes etc claim-, search-, console- .  This way I can then filter by the prefix in device claiming. Thoughts?

pgalic96, 2020-03-19:
I am not sure about this, in CLI there is no constraint on setting entity attributes. I think this should be left to the user, which can set prefixes and organise attributes, and search for them.

mjamescompton, 2020-03-19:
Not necessary but i'm thinking if we have search attributes these will be shown in the form on device claiming, I could instruct user to manually add a prefix to display on device claiming but this would require better docs and would be prone to user errors.

pgalic96, 2020-03-19:
I am fine with having a prefix field, thoughts @bafonins @kschiffer ?

PR #2752 (2020-06-20): Set device attributes in console

#### Summary

Closes #2134 

#### Changes

- Change attributes description to be generic
- Add functionality for setting custom attributes for devices in the General settings

[image]

#### Testing

- Add / Remove device attributes, check actual request in the Network tab of developer console, check Redux state

##### Regressions

- Device general settings form, tested by editing all fields

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API [... cut]
closed 2020-06-20 by pgalic96