TheThingsNetwork/lorawan-stack #2135: Document webhook path variables

adriansmares, 2020-03-11:
#### Summary

The webhook path variables should be explicitly be documented as part of the webhook setup documentation.

#### Why do we need this ?

Users have reported that they would like such a feature (webhook path variables), without knowing that we already implemented this. 

#### What is already there? What do you see now?

The variables are mentioned in the `CHANGELOG` only, no clear usage instructions.

#### What is missing? What do you want to see?

A section about the variables as part of the webhook guide.

#### How do you propose to document this?

Add the section to the guide.

#### Can you do this yourself and submit a Pull Request?

Yes.

benolayinka, 2020-06-22:
@adriansmares the [webhooks docs](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/doc/content/integrations/webhooks/) say `/api/v3/as/applications/{application_id}/webhooks/{webhook_id}/devices/{device_id}/down/push`, is this what you mean?

The changelog documents 
`appID, appEUI, joinEUI, devID, devEUI and devAddr` but I'm not sure how `AppEUI, JoinEUI, DevEUI or DevAddr` would be used in a webhook?

adriansmares, 2020-06-22:
The path you've linked is for scheduling downlink messages. The path variables (`appID`, `appEUI` etc.) are about the paths that you set in the webhook for uplink messages such as the base path, the uplink path or other paths that you may provide. 

You may test this by creating a webhook on a [webhook.site](https://webhook.site) by setting the URL they provide as a base URL and adding the variables to the message path (uplink message path).

Regarding the format used by these variables, check [this](https://en.wikipedia.org/wiki/URL_Template) short article on URL templates. I think the examples in the Wikipedia article are easier to understand than the long RFC itself, but the RFC is linked there as well.

benolayinka, 2020-06-22:
Got it, so the AS posts webhooks like
`https://app.example.com/{base}/{topic}?appID={appID}&appEUI={appEUI}`

That look right?

adriansmares, 2020-06-22:
> 
> 
> Got it, so the AS posts webhooks like
> `https://app.example.com/{base}/{topic}?appID={appID}&appEUI={appEUI}`
> 
> That look right?

It's more like `{base}{/path}` where base is the base URL (which can contain variables itself) and `path` is the path depending on the message type (and can again, contain variables).

So if the base is `http://example.com{/appID}` and the path is `/up{/devID}` an uplink from the device `dev1` of application `app1` will be posted at `http://example.com/app1/up/dev1`

PR #2783 (2020-07-06): Add webhook path variable documentation

#### Summary

Closes #2135 

#### Changes

- Add webhook path variable doc
- Restructure webhook docs

#### Testing

Run locally

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the changes are fully compatible with existing API, database, configuration and CLI.
- [x] Documentation: Relevant documentat [... cut]
closed 2020-07-08 by adriansmares