TheThingsNetwork/lorawan-stack #2139: Inconsistent usage of AppEUI and JoinEUI

laurensslats, 2020-03-12:
#### Summary
While adding a device, the term `JoinEUI` is used. In the **General Settings** page, `AppEUI` is used.

Let's be consistent and use one term. Also, the description of AppEUI is incorrect.

In addition, not everyone is familiar with the term JoinEUI, so this requires clarification.

...

#### Steps to Reproduce
1. Add a device - check the field `JoinEUI`
2. Go to the **General Settings** page of the device, and look for the `AppEUI` (the JoinEUI is not there)

#### What do you see now?

[image]

[image]

...

#### What do you want to see instead?
Add the description: *JoinEUI is the same as the AppEUI*

[image]

In the **General Settings** page, change `AppEUI` to `JoinEUI` and add the description: *JoinEUI is the same as the AppEUI*

[image]

...

#### Environment
v3.6.0

...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

bafonins, 2020-03-12:
We use `AppEUI` and `JoinEUI` interchangeably in the device general settings. `AppEUI` is used for device with `lorawan_version` 1.0.0-1.0.3, `JoinEUI` for 1.0.4+, that is per specification.

According to https://github.com/TheThingsNetwork/lorawan-stack/pull/1616#issuecomment-561234789 `AppEUI identifies the Join Server` is not correct description. I guess we can change it to something like `AppEUI typically identifies an application`.

laurensslats, 2020-03-26:
Ok. Proposing the following changes:

```
The AppEUI is a global application ID that uniquely identifies the owner of the ­device. AppEUI is used for devices with LoRaWAN version 1.0x, JoinEUI for 1.1x
```

[image]

```
AppEUI is used for devices with LoRaWAN version 1.0x, JoinEUI for 1.1x
```

[image]

johanstokking, 2020-03-26:
Technically, in 1.0.4 it's named `JoinEUI` already

bafonins, 2020-03-26:
Yeah, sorry. We display `App EUI` for devices with lw version >1.0.4, and `Join EUI` for 1.0.4+

johanstokking, 2020-03-26:
Yeah the thing is though that we need the LoRaWAN version for that, which is an NS field.

In JS-only, we don't know this

bafonins, 2020-04-02:
So, shall we default to `App EUI` then?

johanstokking, 2020-04-02:
I think we can do both; **AppEUI / JoinEUI**

closed 2020-06-04 by bafonins