TheThingsNetwork/lorawan-stack #2140: Clear list Frequency Plan when adding GWs

laurensslats, 2020-03-12:
#### Summary
Unclear what frequency plan to use when connecting gateways due to options that shouldn't be displayed. 

[image]

...

#### Why do we need this?
Clarify gateway configuration and avoid support questions.

...

#### What is already there? What do you see now?
See img above

...

#### What is missing? What do you want to see?
Remove all the fields that contain (TTN) - these options are only relevant for configuring devices. 
...

#### How do you propose to implement this?
> Filter when there's no `radio`. Fastest is a setting in the index. We need that anyway to avoid fetching the plan to check radio
-- see: https://thethingsindustries.slack.com/archives/C8SCJ2GG1/p1583828355008900

...

laurensslats, 2020-03-12:
See: https://github.com/TheThingsIndustries/lorawan-stack/issues/2001

htdvisser, 2020-03-12:
> Filter when there's no radio.

This will give incorrect results. For example, it would hide `AS_923_925_TTN_AU` which would impact the duty cycle settings of the gateway.

> Fastest is a setting in the index.

Then we would also need to add an extra filter to the API in addition to the `base_frequency` (which isn't used by the console either by the way). I would suggest something like an "applicability" enum where the 0-value is the current behavior (match all), and we have 1 and 2 for gateways and end devices.

kschiffer, 2020-07-02:
> Then we would also need to add an extra filter to the API in addition to the `base_frequency` (which isn't used by the console either by the way). I would suggest something like an "applicability" [...]

Who should pick this up, @htdvisser ? I can then take care of using that in the Console.

htdvisser, 2020-07-02:
I don't know if we should still do this. The only two frequency plans that are currently still equivalent for Gateways and End Devices are `EU_863_870` and `EU_863_870_TTN`. Those are now displayed as `Europe 863-870 MHz (SF12 for RX2)` and `Europe 863-870 MHz (SF9 for RX2 - recommended)`.

I'm afraid that hiding `EU_863_870_TTN` (SF12) would be confusing for users who do want to use SF12.

Do we really still need this, @laurensslats?

johanstokking, 2020-07-17:
Ping @laurensslats

laurensslats, 2020-07-17:
Let's snooze this for now. 

When we implement this, we also need to change the name, as it should only display `Europe EU863-870 MHz`

closed 2020-07-17 by laurensslats