TheThingsNetwork/lorawan-stack #2170: Fix random redirects to the overview page

kschiffer, 2020-03-17:
#### Summary
The console will sometimes randomly redirect the user back to the console overview page. This is likely caused by a race condition in our feature toggle logic.

#### Steps to Reproduce
Unfortunately hard to reproduce as this seems to occur arbitrarily, but it happens when entering a view that is restricted via the `withFeatureRequirement` HOC.

#### What do you see now?
Random redirects to the overview page, that would otherwise occur when a user has no permission to view a page.

#### What do you want to see instead?
Such faulty redirect to not occur.

#### Environment
v3.6.1

#### How do you propose to implement this?
I have to do some investigation to understand why this happens, but I think it's pretty likely that his is caused by a race condition (the feature requirement HOC being rendered with old right props)

#### Can you do this yourself and submit a Pull Request?
Yes.

kschiffer, 2020-09-16:
After trying to reproduce for a while I could not succeed. Adding `missing details` for the time being. If anyone witnesses this, please try to drill down the reproduction steps.

johanstokking, 2020-09-17:
Anyone please reopen this issue when it occurs and paste details.

@kschiffer what details do you need? Browser console log?

closed 2020-09-17 by johanstokking

kschiffer, 2020-09-17:
This likely happens with collaborators that have fewer rights or when certain stack components are disabled, so that info would be good to know.