TheThingsNetwork/lorawan-stack #2177: Support AsNs downlink queue operations on either of the queues

rvolosatovs, 2020-03-17:
#### Summary

Support AsNs downlink queue operations on either of the queues

#### Why do we need this?

To allow AS to efficiently manipulate the queue

#### What is already there? What do you see now?

Queue operations are performed on both current and pending session queues

#### What is missing? What do you want to see?

Ability to select the session queue to perform the operation on

#### How do you propose to implement this?

1. Rename `DownlinkQueueRequest` -> `AppAsDownlinkQueueRequest`
2. Introduce `AsNsDownlinkQueueRequest` with:
   - device IDs
   - 2 messages containing the queue (`pending_downlinks` and `current_downlinks` or similar naming)

(2.) breaks internal API, which is fine for us, since we're going to update the stack atomically, however do we expose NS API to our users? If so, we'd need to follow a different approach(with enums and additional field)

#### Can you do this yourself and submit a Pull Request?

Yes

rvolosatovs, 2020-03-17:
cc @johanstokking @adriansmares

rvolosatovs, 2020-03-17:
Blocking #2163

johanstokking, 2020-03-17:
The [compatibility commitment](https://github.com/thethingsnetwork/lorawan-stack#commitments-and-releases) is to be hosted a bit vague about "towards applications". Technically, AS is application-layer, so NS to AS = NS towards applications. In that sense, we cannot really change this. @htdvisser do you agree or do you think we can break NS-AS API in major?

> #### Why do we need this?
> To allow AS to efficiently manipulate the queue

Can you clarify this? Why can't the AS specify the session by the session key ID per item?

rvolosatovs, 2020-03-17:
We don't really need this in fact, because AS always lists the queue anyway and should be smart not to recompute stuff it does not need to recompute.
The only inefficiency here is that AS needs to send more items, that it actually needs to (e.g. if only one session queue needs to be replaced), but that's not worth the optimization at this point of time.

htdvisser, 2020-03-18:
For the record, an API change in the NS-AS API falls under

> We will not break the API between components and events within minor versions. So at least the same minor versions of components are compatible with each other

rvolosatovs, 2021-01-19:
This is now supported

closed 2021-01-19 by rvolosatovs