TheThingsNetwork/lorawan-stack #2178: Error when executing simulate command of the TTN Cli tool

cheribhai, 2020-03-17:
#### Summary
Simulate command of the TTN Cli tool gives an error

#### Steps to Reproduce
ttn-lw-cli simulate uplink \
    --dev-addr <dev addr> \
    --frm-payload <payload> \
    --lorawan-version 1.0.2 \
    --log.level debug \
    --gateway-id <gateway id> \
    --f-nwk-s-int-key <network session int key> \   
    --app-s-key <app session key> \
    --lorawan-phy-version 1.0.2-b \
    --band-id EU_863_870 

#### What do you see now?
Command returns an error and no message is received on the TTN Device Data console ...
DEBUG Using access token (valid until 10:39AM)
DEBUG ccResolverWrapper: sending update to cc: {[{eywa-networks.eu1.cloud.thethings.industries:8884  <nil> 0 <nil>}] <nil> <nil>}
DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc0002a8260, {CONNECTING <nil>}
DEBUG pickfirstBalancer: HandleSubConnStateChange: 0xc0002a8260, {READY <nil>}
INFO Sent uplink                            
DEBUG Finished streaming call                  duration=20.000540346s error=context deadline exceeded grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc
error:unknown:unknown (context deadline exceeded)

#### What do you want to see instead?
Command executes as expected and the message is seen in the TTn Device Data console.

#### Environment
: ttn-lw-cli version                
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.6.1
Build date:          2020-03-13T11:26:42Z
Git commit:          8be68232e
Go version:          go1.13.8
OS/Arch:             linux/amd64

johanstokking, 2020-03-19:
@cheribhai please provide detailed steps to reproduce.

- how did you create the gateway?
- how did you create the device?
- what exactly is the command for simulate?

Note that the `simulate` command is hidden because it's highly experimental.

cheribhai, 2020-03-19:
@johanstokking I have created the device (via ABP) and the gateway with TTN generated device address and keys.
The device and gateway have been created purely for simulation purpose and are not connected to any physical device
I have provided the ttn simulate command in the description (minus sensitive info)

johanstokking, 2020-03-19:
OK, but please copy the exact commands so we can reproduce the issue.

Also, what do the server logs say, with debug logs enabled?

cheribhai, 2020-03-20:
@johanstokking I have updated the 'Steps to reproduce' section with a bash script which contains all the details and commands to create the application, gateway, device and then run the simulate command.
We are not self-hosting the TTN V3 stack. We have (last week) subscribed to the TTN hosted V3 (Industries) stack, so I'm not really sure If I can get to the server logs myself.
I do need to run some end to end tests before migrating the applications / devices from V2 to V3, so if there is any other option to simulate (maybe direct api calls) please advise.

johanstokking, 2020-03-20:
OK I see.

Please specify frequency, bandwidth and spreading factor. Since you didn't specify the frequency plan as part of the device create, only the three default channels are enabled (so `868100000`, `868300000` and `868500000`), bandwidth is `125000` and spreading factor like `7`. You also need to specify `f-cnt` (incrementing).

Finally, you can subscribe to gateway events (`ttn-lw-cli events subscribe --gateway-id $GTW_ID`)

cheribhai, 2020-03-20:
As can be seen in the description, I am specifying the frequency plan in the request.
I can confirm this with the call
t```
tn-lw-cli end-devices get $APP_ID $DEV_ID --frequency-plan-id
{
  "ids": {
    "device_id": "dev-sim-2178",
    "application_ids": {
      "application_id": "app-sim-2178-3"
    },
    "dev_addr": "00E4304D"
  },
  "created_at": "2020-03-20T17:46:14.357Z",
  "updated_at": "2020-03-20T17:46:45.514700632Z",
  "network_server_address": "<domain>.cloud.thethings.industries",
  "frequency_plan_id": "EU_863_870"
}

```
I had made a mistake with,
`--session.dev-addr $DEV_ID \`
which I have corrected to 
`--session.dev-addr $DEV_ADDR \`

Running the simulate uplink with explicitly specifying the frequency, bandwith and f-cnt still gives the same error,
```
ttn-lw-cli simulate uplink \
    --dev-addr $DEV_ADDR \
    --frm-payload 4869 \
    --lorawan-version $LORAWAN_VERSION \
    --log.level debug \
    --gateway-id $GTW_ID \
    --f-nwk-s-int-key $NWK_S_KEY \
    --app-s-key $APP_S_KEY \
    --lorawan-phy-version $LORAWAN_PHY_VERSION \
    --band-id $FREQ_PLAN \
    --bandwidth 125000 \
    --frequency 868500000 \
    --spreading-factor 7 \
    --f-cnt 1
```

cheribhai, 2020-03-20:
The gateway events show the error ..

```
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
    "namespace": "pkg/gatewayserver",
    "name": "host_handle",
    "message_format": "host `{host}` failed to handle message",
    "attributes": {
        "host": "cluster"
      },
    "cause": {
      "namespace": "pkg/networkserver",
      "name": "device_not_found",
      "message_format": "device not found",
      "correlation_id": "b643b1da80814b4fa81b438625abd73c",
      "code": 5
    },
    "code": 5
  }
```

johanstokking, 2020-03-21:
> As can be seen in the description, I am specifying the frequency plan in the create device request.

Right, that wasn't clear on my side. I meant that with ABP, you didn't specify the enabled channels. Those are the "factory reset frequencies". If you don't specify those, only 868.1, 868.3 and 868.5 are enabled.

In any case, you can either specify `frequency` or `channel-index`. If you don't specify either, `channel-index` is 0, and 868.1 as channel 0 is used. Also, you can either specify `data-rate-index` or `spreading-factor`/`bandwidth`. If you don't specify either, data rate 0 is used, which is SF12BW125. So that's not the issue.

I cannot reproduce your issue locally with TTS. It works immediately. What I had to fix though, is to use `$DEV_ADDR` instead of `$DEV_ID` in your example.

Since the NS cannot find the device, I suppose the problem is the (reused) frame counter. Please make sure you increase that counter on any attempt. You may want to set `--mac.settings.resets-f-cnt` to `true` for testing.

What we have currently running in The Things Industries Cloud Hosted is 3.6.1. We patched some uplink related issues in 3.7.0-rc1, which we're deploying Monday.

cheribhai, 2020-03-21:
@johanstokking Incrementing the `f-cnt` counter was what I was missing (my bad). Sorry about that - I should have seen this.

But thanks a ton! It works now and is really helpful - I will be setting `--mac.settings.resets-f-cnt` to `true` for devices that will only be used for simulation.

I see another (random) error, but I have opened a separate ticket for this - https://github.com/TheThingsNetwork/lorawan-stack/issues/2210

closed 2020-03-24 by johanstokking