TheThingsNetwork/lorawan-stack #2227: Add Auth Method + ID and Remote IP to CRUD event data

htdvisser, 2020-03-24:
#### Summary

We should add the Auth Method and API Key / Access Token ID to CRUD events. It would also be good to add the remote IP.

#### Why do we need this?

For audit logging

#### What is missing? What do you want to see?

We should extract both the auth (rpcmetadata) and remote address (peer) from the context when creating an event.

#### How do you propose to implement this?

- Add fields to the Event definition (api)
- In `events.New`, set those fields.
- Can use `peer.FromContext` to extract the remote IP. Propbably also need to consider `x-forwarded-for` headers.
- Can use `rpcmetadata.FromIncomingContext` to extract the `AuthType` (usually `Bearer`) and `AuthValue`. Use `auth.SplitToken` on the latter to extract the `tokenType` and `tokenID`. The `tokenKey` should **NEVER** be added to events.

htdvisser, 2020-03-24:
I think this is relatively easy (with the instructions above), so if anyone else wants to take over, let me know.

neoaggelos, 2020-05-21:
Picking this one up

PR #2580 (2020-06-03): Add authentication method, ID and IP event metadata when available

#### Summary

Closes #2227

#### Changes

- Add Auth Method, Token ID, Remote IP as event metadata when available
- Adds `rpcmetadata.XForwardedFor` field

#### Testing

Updated unit tests, tested in development environment with dummy API create + delete events.

##### Regressions

...

#### Notes for Reviewers

- Cannot use `auth.SplitToken()` since it creates an import cycle in the tests.
- We might want to do something different for `x-forwarded-for`.

#### Checklist

- [X] Scope: The referenced issue is addressed, there ar [... cut]
closed 2020-06-03 by neoaggelos