TheThingsNetwork/lorawan-stack #2230: Console feature gaps v2/v3

kschiffer, 2020-03-25:
#### Summary
We organize our work towards closing the current feature gap of the v2 and v3 console. This umbrella issue keeps track of the progress.

#### Why do we need this?
- We should not work on completely new features until the v3 console can provide all features of v2 console.
- Organizing our process / roadmap

#### What is already there? What do you see now?
We already provide most of the features that users know from v2, but there are still some gaps.

#### What is missing? What do you want to see?
##### General
- [ ] Profile settings (Name, Description, Email, Profile Picture, Delete own account) ⟶ #1422
- [ ] Providing Japanese and Chinese language version of the console ⟶ Issue follows
- [ ] Simple filters for event streams ⟶ Issue follows

##### Applications
- [ ] Providing advance payload format converter and validator ⟶ Issue follows
- [ ] Providing payload format testing (dry-running byte payloads through the payload formatted) ⟶ Issue follows
- [ ] Providing template tiles for webhook integrations ⟶ #945 

##### End Devices
- [ ] Showing end device statuses in end device list ⟶ Issue follows
- [ ] Showing the Device EUIs in the end device list ⟶ Issue follows
- [ ] Showing end device last seen and frame counts in device overview ⟶ #2167 
- [ ] Resetting frame counts for end devices ⟶ Issue follows
- [ ] Scheduling downlink ⟶ Issue follows
- [ ] Setting frame counter widths ⟶ Issue follows

##### Gateways
- [ ] Setting and displaying gateway manufacturer info (Brand, Model, Antenna(s), etc) ⟶ Issue follows
- [ ] Setting automatic update behavior ⟶ Issue follows
- [ ] Setting Indoor/Outdoor location flag ⟶ Issue follows

@htdvisser @johanstokking please review the list. There might be things in there that we actually don't want to add to v3, or in different form.

johanstokking, 2020-03-25:
I'm very happy with the overview issue. And I think it's good to have the issues filed. We can then prioritize them and put them in the concerning milestones.

kschiffer, 2020-04-29:
I filed all issues now. @laurensslats any specific priority you want us to respect?

benolayinka, 2020-07-09:
@kschiffer how about creating users in the console? This is still CLI only

kschiffer, 2020-07-14:
Yes but this is not a v2/v3 feature app. You can open a separate issue for this so we can track this.

kschiffer, 2020-09-16:
I'll move this to the Backlog milestone since the individual issues are already prioritized correctly and this issue is more to keep track overall.

johanstokking, 2021-02-08:
I think it's time to close this, especially because we have individual issues that can be tracked.

closed 2021-02-08 by johanstokking