TheThingsNetwork/lorawan-stack #2231: Event stream filters in console

kschiffer, 2020-03-25:
#### Summary
We should provide filtering features for our data view / event stream.

#### Why do we need this?
- Closing v2 feature gap
- Supporting advance use cases for data monitoring

#### What is already there? What do you see now?
Event streaming for all our entities.

#### What is missing? What do you want to see?
Filtering options to improve utility of the event stream

#### Environment

...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

kschiffer, 2021-02-15:
With regards to [https://github.com/TheThingsNetwork/lorawan-stack/issues/3713#issuecomment-779297814,](https://github.com/TheThingsNetwork/lorawan-stack/issues/3713#issuecomment-779297814,) we need to expand the functionality here a bit. There are scenarios where applications have hundreds or thousands of end-devices. This will crash the Console altogether when the event view is open. Hence, it is necessary to enforce some filter for heavy-load situations.

Such filters could be:

*   Limit the entity scope for application event streams to max 20 (or so) end devices
*   Only show the `forward` event type for uplinks downlinks and similar outfiltering of auxiliary event

We could use different levels of enforced filtering and use it for some predefined "events per second"-thresholds.

Ideas welcome cc @bafonins @htdvisser @johanstokking

htdvisser, 2021-02-16:
The number of registered end devices isn't necessarily an indication of the amount of traffic, so I don't think it would be a good idea to do different things depending on the number of registered end devices.

My suggestion would be to extend the API to allow filtering on event name. Every events view in the Console would then give the user some UI where they can check/uncheck events they're interested in. The event names that are selected by default can then be adapted to show the most relevant events for a given entity.

kschiffer, 2021-02-16:
Blocked on #3817

kschiffer, 2021-05-06:
I had originally planned to fix this with a general overhaul of the event stream via #3817, but I think it's necessary to make some quick adjustments now to make the event stream less verbose. A simple solution here is to add a verbose switch to the event stream, which will enable users to filter out many events that only add noise to the event output.

I propose the following list of events to be shown by default, when verbosity is deactivated:

```
as.down.data.decode.fail
as.down.data.encode.fail
as.down.data.forward
as.packages.loraclouddmsv1.fail
as.packages.loracloudglsv3.fail
as.pubsub.fail
as.up.data.decode.fail
as.up.data.forward
gs.down.send
gs.down.tx.fail
gs.gateway.connect
gs.gateway.disconnect
gs.status.fail
gs.status.forward
gs.up.fail
gs.up.forward
js.join.accept
js.join.reject
ns.down.data.schedule.fail
ns.down.join.schedule.fail
ns.up.join.cluster.fail
ns.up.join.interop.fail
organization.*
user.*
gateway.*
application.*
end_device.*
```

I've selected these events based on what is likely of direct interest to users. This includes

*   any failure event
*   one event per received data message, which is the `as.up/down.data.forward`
*   one event per join request/reject
*   one event per received message for gateways
*   gateway metrics (though i find that these are often sent many times repeatedly)
*   any entity registry event (these are usually not very frequent)

Please review this list @htdvisser @johanstokking @neoaggelos @adriansmares, so I can finalize the list and coordinate implementation.

neoaggelos, 2021-05-06:
The list of events LGTM in general with one exception: I would swap `gs.status.forward` with `gs.status.receive`, since the status messages are important for the GS, not the upstream hosts (at the moment at least). This also gives us 1 instead of 2 events per status message (we get one `gs.status.forward` per upstream host).

htdvisser, 2021-05-06:
I think it would also be useful to have these:

- `*.end_device.*` -- end device updates on the NS/AS/JS are mostly useful when we also have history, because then you can compare behavior before/after the update.
- `as.{up,down}.data.decode.warning` -- we shouldn't ignore warnings
- `as.{up,down}.data.drop` -- dropped uplinks/downlinks in the AS are almost always abnormal
- `as.up.join.drop` -- neither are dropped joins in the AS
- `as.down.data.queue.invalid`/`as.down.data.queue.lost` (maybe, @adriansmares?)
- `*.api-key.*` -- these are security events
- `*.collaborator.*` -- these too
- `ns.mac.*.answer.reject` -- these indicate a problem in the MAC state, shouldn't be hidden
- `ns.up.data.process` -- this is the first event that contains a de-duplicated uplink

adriansmares, 2021-05-06:
>     * `as.down.data.queue.invalid`/`as.down.data.queue.lost` (maybe, @adriansmares?)

These events actually are no longer possible (:tada:) and we should deprecate them. Thanks for reminding me.

htdvisser, 2021-05-07:
Assuming that the majority of users will not use verbose mode, _they will_ ignore warnings if we don't show them. 

I'm fine with hiding "expected" things that are not relevant to most users. Warnings are not "expected" and are relevant, because users will probably need to take action on them.

kschiffer, 2021-05-07:
Here's the updated list:

```
as.up.join.drop
as.up.data.drop
as.up.down.drop
as.up.data.decode.warning
as.down.data.decode.fail
as.down.data.encode.fail
as.down.data.decode.warning
as.down.data.forward
as.packages.loraclouddmsv1.fail
as.packages.loracloudglsv3.fail
as.pubsub.fail
as.up.data.decode.fail
as.up.data.forward
gs.down.send
gs.down.tx.fail
gs.gateway.connect
gs.gateway.disconnect
gs.status.fail
gs.status.receive
gs.up.fail
gs.up.receive
js.join.accept
js.join.reject
ns.down.data.schedule.fail
ns.down.join.schedule.fail
ns.up.join.cluster.fail
ns.up.join.interop.fail
ns.mac.beacon_freq.answer.reject
ns.mac.dl_channel.answer.reject
ns.mac.link_adr.answer.reject
ns.mac.new_channel.answer.reject
ns.mac.rx_param_setup.answer.reject
organization.*
user.*
gateway.*
application.*
end_device.*
client.*
oauth.*
```

johanstokking, 2021-05-07:
@kschiffer with a simple filter to start with, I triaged this issue. Please check, change labels, assignees and milestone if needed, and remove `needs/triage`.

kschiffer, 2021-05-07:
Yes, I'll take on the verbosity filter. Shouldn't be a big deal.

htdvisser, 2021-09-23:
As discussed with @kschiffer, we'll add support for filtering events by name on the backend. 

I'll pick up the API and backend implementation after which @kschiffer will update the Console.

htdvisser, 2021-09-27:
@kschiffer I just pushed the backend work to [feature/2231-event-filter](https://github.com/TheThingsNetwork/lorawan-stack/compare/feature/2231-event-filter).

I can imagine that for the frontend it can be useful to know the full list of event names. In `pkg/events/grpc/grpc.go` you'll see how the EventsServer builds that list. We could write a little tool to extract the names into a JS/JSON file (similar to the one that creates `messages.json`).

htdvisser, 2021-11-15:
@kschiffer did you still want to implement the rest of this or did you only want to pick up the part in #4831?

kschiffer, 2021-11-16:
I can stay assigned here. I've removed the milestone for now.

kschiffer, 2022-06-29:
Yes. Let's schedule this and aim for completion in Q3.

KrishnaIyer, 2024-10-14:
We have basic filtering of events (verbose). The proposal here is great if we have a General "events" page on the console and we can then do some filtering there. If we have such a page, then indeed we can look at this generic filtering system.

However, the way most users (including me) use the console is to navigate to the target gateway, application or end device and look into the events there. Here, we can make a few informed choices about the filtering in each of those pages (contexts). That is handled here: https://github.com/TheThingsNetwork/lorawan-stack/issues/7342

closed 2024-10-14 by KrishnaIyer