TheThingsNetwork/lorawan-stack #2232: Payload format testing

kschiffer, 2020-03-25:
#### Summary
We should provide testing functionality for our payload format settings in the Console.

#### Why do we need this?
- to facilitate debugging and development for our users
- to close v2/v3 feature gap

#### What is already there? What do you see now?
Payload format settings views for application and end device level.

#### What is missing? What do you want to see?
Functionality similar to the v2 payload format testing:
[image]
I think we can basically do a 1:1 adaption of this.

#### Environment
v3.7.0

#### How do you propose to implement this?
1. Add backend API to test set payload formatters (at least that's how it currently works in v2)
2. Update JS SDK
3. Extend payload format container component to use testing capabilities

#### Can you do this yourself and submit a Pull Request?
I can do the frontend part. Who can pick up the backend part? cc @htdvisser Please assign accordingly.

johanstokking, 2020-03-25:
> * Add backend API to test set payload formatters (at least that's how it currently works in v2)

This is indeed how it needs to work, with RPCs in `AppAs` to execute the provided script for encoding and decoding, according to the type etc. @adriansmares please have a look at `pkg/messageprocessors/payload.go`. We currently do have a GRPC service there that we don't use (`payload_grpc.go`), which is intended to wrap an encoder/decoder. We need something like this, but also specify which type of encoder/decoder we want, i.e. what the user selected.

One relevant change from V2 is that we are considering the end device version identifiers here, because those may affect how payload is to be decoded. That needs to be reflected in the Console.

kschiffer, 2020-05-19:
@adriansmares please reassign me once you're finished with your part.

laurensslats, 2020-06-17:
Blocked by #2670

bafonins, 2020-10-04:
When testing formatters for uplink messages it all comes down to checking for `as.up.data.forward`, `as.up.data.decode.warning` and `as.up.data.decode.fail` events.

1. In case of user defined warnings (in the `warnings` entry of the js formatter) the `as.up.data.decode.warning` event gets sent containing decoded payload (`decoded_payload`) and at least one warning (`decoded_payload_warnings`).
2. In case of user defined errors (in the `errors` entry of the js formatte) or runtime exceptions while executing the uplink payload formatter the `as.up.data.decode.fail` event gets sent containing an error in the `data` entry of the event.
3. Otherwise only the `as.up.data.foraward` gets send with the `decoded_payload` field.

I wasnt able to find how things are for downlinks with custom (valid and invalid) js payload formatters w/ and w/o the actual end device. @adriansmares Could you check that out or describe the steps on how I can reproduce warning and error events being dispatched for downlinks?

johanstokking, 2020-10-05:
We need API for this in AS, so you can use rpcs to test directly without using events. The response of those rpcs would indeed include any warnings or errors produced by the payload formatter.

bafonins, 2020-10-05:
A dedicated rpc will definitely make things easier for us. Is there an issue for this already?

johanstokking, 2020-10-06:
It's this issue, [see above](https://github.com/TheThingsNetwork/lorawan-stack/issues/2232#issuecomment-604009221) with some tips.

Let me know if you need anything from me by assigning me.

bafonins, 2020-10-06:
Assigning @adriansmares for adding AS support

kschiffer, 2021-02-23:
@adriansmares do you have an ETA here?

adriansmares, 2021-02-23:
> @adriansmares do you have an ETA here?

This is in `Next Up`, so indefinite unless this gets scheduled.

kschiffer, 2021-02-23:
@htdvisser @johanstokking please consider raising the priority of this since it has been requested various times. See TheThingsIndustries/lorawan-stack-support#252

closed 2021-04-12 by bafonins