TheThingsNetwork/lorawan-stack #2234: Simulate uplinks in the console

kschiffer, 2020-03-25:
#### Summary
We should provide functionality to simulate uplinks for end devices in the console.

#### Why do we need this?
- Helps users debugging their apps/end devices
- Close v2/v3 feature gap

#### What is already there? What do you see now?
Nothing of that sort. See also #2232 

#### What is missing? What do you want to see?
UI to simulate uplinks on end device level, see v2:
[image]

#### Environment
v3.7.0

#### How do you propose to implement this?
Details to follow.

#### Can you do this yourself and submit a Pull Request?
Yes but anyone feel free to pick up.

johanstokking, 2020-03-25:
This needs `AppAs` support for a dry uplink.

@adriansmares in V2 this is quite hacky; afaik we generate some random metadata. What's most important is that we run the user specified `frm_payload` through the decoders and it becomes available like any other data uplink message. The purpose is to test decoders and integrations primarily.

laurensslats, 2020-06-11:
This is a much requested feature. Let's try focussing on this issue by the end of June / early July.

_The Docs-team is now starting with documentation creation on writing payload formatters._

kschiffer, 2020-06-19:
@adriansmares please reassign me once the `AppAs` support is done. Happy to pick this up directly thereafter.

adriansmares, 2020-07-17:
> @adriansmares please reassign me once the `AppAs` support is done. Happy to pick this up directly thereafter.

The API has been added in #2919

closed 2020-09-25 by bafonins