TheThingsNetwork/lorawan-stack #2238: Improve event handling in Network Server

rvolosatovs, 2020-03-25:
#### Summary

Improve event handling in Network Server
Refs #2167 

#### Why do we need this?

For easier debugging

#### What is already there? What do you see now?

Several events, but we can do better

#### What is missing? What do you want to see?

- [ ] Uplink event improvements
	- Trigger an event when uplink is received
	- Trigger an event when uplink is successfully processed
	- Trigger an event when NS sends the join-request to *a* JS
	- Trigger an event when NS receives a response from JS
	- Trigger an event when NS sends an application uplink to AS

- [ ] Downlink event improvements
	- Trigger an event when errors from GS are received including errors per-gateway

#### How do you propose to implement this?

Add events and handling

#### Can you do this yourself and submit a Pull Request?

yes

johanstokking, 2020-03-25:
cc @kschiffer @bafonins

bafonins, 2020-03-25:
Sounds good.

> Trigger an event when uplink is received

Would be nice if this event has a different description from `as.up.data.receive`, namely `receive uplink data message` as it might be confusing for the users to get several of those. We currently do not distinguish events dispatched by as/ns anyhow in the Console.
I think the only option to do this is by checking the `name` field of the event. @kschiffer @johanstokking do you think it is worth adding the source of the event (stack component)? It could be an icon/abbreviation. We can elaborate that in https://github.com/TheThingsNetwork/lorawan-stack/issues/1962

rvolosatovs, 2020-03-25:
> Sounds good.
>
>
> Would be nice if this event has a different description from `as.up.data.receive`, namely `receive uplink data message` as it might be confusing for the users to get several of those. [...]

Good one, I already changed this for forwarding in https://github.com/TheThingsNetwork/lorawan-stack/pull/2221, so the event will now state that the respective message is forwarded to AS, don't think we should change the description of NS events though.
I think having a small box with abbreviated component name (e.g. `NS`, `AS, etc.) on each event would be extremely helpful. You could, in fact, just take the string before first `.`, make it uppercase and print in this "box". The first part of event name always indicates the component, where it comes from.

PR #2221 (2020-04-09): Network Server event handling and monitoring

#### Summary

Fix NS uplink event handling and monitoring, event testing assertions
Closes #2238

#### Changes

- Refactor NS event handling
	- On uplink, trigger event on:
		- receive
		- successful processing
		- join-request sending to JS
		- join response receival from JS
	- On downlink, trigger event on:
		- sending TxRequest to GS
		- failure of TxRequest to GS
		- success of TxRequest to GS

- Only register and trigger AS "forwarding" events when actually sending to AS
- Significantly refactor flow test events
- Add 1.1 Class C OTAA flow tes [... cut]
closed 2020-04-09 by rvolosatovs