TheThingsNetwork/lorawan-stack #2242: Issue with redis, all uplinks are dropped

alexbn71, 2020-03-26:
#### Summary
The newest versions has an issue related with redis that drops all uplinks messages received by nodes.

#### Steps to Reproduce
I use docker-compose.yml with this settings for redis:

`  redis:
    image: redis
    command: redis-server --appendonly yes
    restart: unless-stopped
    volumes:
      - ${DEV_DATA_DIR:-.env/data}/redis:/data
    ports:
      - "127.0.0.1:6379:6379"

...
...
...

   environment:
      - TTN_LW_REDIS_ADDRESS=redis:6379
`

#### What do you see now?

dropped uplinks in console and this message in the stack:

WARN Finished unary call                      duration=14.29155ms error=error:pkg/redis:store (store error) error_cause=dial tcp 127.0.0.1:6379: connect: connection refused error_correlation_id=91199bf942f5474f947218d76407fab5 error_name=store error_namespace=pkg/redis grpc_code=Unknown grpc_method=HandleUplink grpc_service=ttn.lorawan.v3.GsNs namespace=grpc request_id=01E46M0W9GGPMWS1FE0G6JHP55
#### What do you want to see instead?

#### Environment

the issue appear after an upgrade to 3.6.1 or 3.6.2. 
3.6.0 works well

KrishnaIyer, 2020-03-26:
@alexbn71 : Is this happening for `v3.6.2` as well? What steps did you take to update?

alexbn71, 2020-03-26:
with a _docker-compose pull_ after this change in my docker-compose.yml:

```
  stack:
    image: thethingsnetwork/lorawan-stack:3.6.2
```

KrishnaIyer, 2020-03-26:
Ok I see the issue. Will be fixed in `v3.6.3`. Also you can update to our pre-release new minor version `v3.7.0-rc2` where this is resolved.

alexbn71, 2020-03-26:
Thank you @KrishnaIyer

alexbn71, 2020-03-26:
does this update require a db migration?

KrishnaIyer, 2020-03-26:
If you update to `v3.7.0-rc2` yes. It does. You can also wait for `v3.6.3` which doesn't need one.

alexbn71, 2020-03-26:
and if I migrate the db can I still back to v3.6.0 in case the issue is still there?

KrishnaIyer, 2020-03-26:
Yes. Downgrading won't be a problem.

alexbn71, 2020-03-26:
Upgraded to v3.7.0-rc2 and I can confirm that it works now! Thanks.

closed 2020-03-26 by johanstokking