TheThingsNetwork/lorawan-stack #2247: CLI reads all available config files

neoaggelos, 2020-03-27:
#### Summary

The cli searchs for config files on a number of different locations (local, home dir, config dir). Even though this order suggests an hierarchy where the local files are given precedence, this is not the case. The cli will attempt to read all available files by default, overriding config in a not apparent manner.

#### Steps to Reproduce

1. Create a config file in user directory (`~/.ttn-lw-cli.yaml`) for host `other.example.com`
2. Create a config file in local directory (`./.ttn-lw-cli.yaml`) for host `this.example.com`
3. Run cli

#### What do you see now?

The CLI will try to connect to `other.example.com`, even though there is a local config file with a different host.

#### What do you want to see instead?

The CLI connecting to `this.example.com`, i.e. the host that is defined on the local config file.

This makes having a local config file pointless, since the one from the user directory will always be read instead.

An obvious work-around for this is to use `ttn-lw-cli -c .ttn-lw-cli.yaml`, but this is not obvious and may cause confusion to new users.

#### Environment

3.7.0

#### How do you propose to implement this?

When loading config files, stop after successfully loading one, insteading of looping for the rest of the config files.

#### Can you do this yourself and submit a Pull Request?

Yes

rvolosatovs, 2020-03-27:
I think we should support reading several config files like we do now, however they should have priority. The config in local directory should override more general settings from user config directory.

johanstokking, 2020-03-27:
> This makes having a local config file pointless, since the one from the user directory will always be read instead.

So it seems that order of loading config files is more of a problem, than that there's config merging in general?

As stated in #2248, not merging configuration breaks the compatibility commitment (https://github.com/thethingsnetwork/lorawan-stack#commitments-and-releases point 4). That compatibility commitment is holy and we do not allow exceptions other than point 6.

neoaggelos, 2020-03-29:
> So it seems that order of loading config files is more of a problem, than that there's config merging in general?

Indeed. Maybe reversing the order that the config files are loaded makes more sense. That way, local files will be read last (if present) and thus have higher priority.

But that also changes existing behaviour, so it's not a "better" solution per se.

johanstokking, 2020-03-30:
> That way, local files will be read last (if present) and thus have higher priority.

OK so if we are overriding local settings from global config files, then that is buggy behavior that we should fix.

So let's stick with merging for compatibility, but start with global and work to local.

PR #2248 (2020-04-01): cli: Local config takes precedence over global

#### Summary

Closes #2247

#### Changes

- Read config files in reverse order (global -> local), so that configuration in local files takes precedence.

#### Notes for Reviewers

~Before reviewing or merging this, please make sure #2247 is actually classified as a bug and not desired behaviour.~

Changed PR after discussion in #2247

Also updating tests.

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The changes are backwards compatible with existing API, database and configu [... cut]
closed 2020-04-01 by neoaggelos