TheThingsNetwork/lorawan-stack #2254: Referring to documentation from within the console

kschiffer, 2020-03-27:
#### Summary
We need to be able to refer to our documentation from within the console.

#### Why do we need this?
Provide help right where it is needed.

#### What is already there? What do you see now?
Documentation

#### What is missing? What do you want to see?
A safe way to refer to the documentation (linking) from within the console.

#### How do you propose to implement this?
- Extend the console configuration to provide a base path to the documentation
- Implement a `<DocLink />` component that will concatenate the base path with the documentation path

#### Can you do this yourself and submit a Pull Request?
Yes. cc @pgalic96 @mjamescompton  feel free to pick up.

kschiffer, 2020-04-14:
I envision a usage like this
```jsx
<DocLink ref="/concepts/webhook-templates">
  <Message content={sharedMessages.webhookTemplattes} />
</DocLink>
```

But we should also provide utilities to generate the documentation link outside of components:
```js
const deviceAddrDocLink = docLink('/concepts/webhook-templates')
```

I think the usage of those will mostly revolve around form titles and descriptions, so it's good to think about how we can easily add documentation links in there.

pgalic96, 2020-04-29:
closed via #2397

PR #2397 (2020-04-29): Link documentation from console

#### Summary

Closes #2254 

#### Changes

- Pass documentation url to the console
- Introduce `DocLink` component which appends `path` to the documentation url
- Introduce `generateDocUrl` util for generating the documentation url

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [x] Testing: The changes are covered with unit tests. The changes are tested [... cut]
closed 2020-04-29 by pgalic96