TheThingsNetwork/lorawan-stack #2258: Inconsistent mac_settings enums values in NS responses

bafonins, 2020-03-29:
#### Summary

References: https://github.com/TheThingsNetwork/lorawan-stack/issues/2047

Values returned by NS for enums within the `mac_settings` field are inconsistent. Some for some NS returns raw values and for others a string representation is returned.

#### Steps to Reproduce

1. Set `mac_settings.rx1_delay.value`

```
// request
{
   "end_device":{
      "mac_settings":{
         "rx1_delay":{
            "value":5 // or 'RX_DELAY_5'
         }
      }
   },
   "field_mask":{
      "paths":[
         "mac_settings.rx1_delay.value",
      ]
   }
}

// response
{
   ...
   "mac_settings":{
      "rx1_delay":{
         "value":5
      },
   }
}

// Note, regardless of what you send as the value for mac_settings.rx1_delay.value you always get the raw enum value.
```

2. Set `mac_settings.ping_slot_periodicity.value`
```
// request

{
   "end_device":{
      "mac_settings":{
         "ping_slot_periodicity":{
            "value": 1 // or 'PING_EVERY_2S'
         }
      }
   },
   "field_mask":{
      "paths":[
         "mac_settings.ping_slot_periodicity.value"
      ]
   }
}

// response

{
   "mac_settings":{
      "ping_slot_periodicity":{
         "value":"PING_EVERY_2S"
      }
   }
}

// Again, regardless of what you send you always get the string representation of the value.

```

#### What do you see now?

Inconsistent values for `mac_settings` enum fields

#### What do you want to see instead?

**All** values should be returned in the same format (raw/string)

#### Environment

v3.7.0-rc2

#### How do you propose to implement this?

I guess first we have to agree on the format. Then it is the matter of adjusting https://github.com/TheThingsNetwork/lorawan-stack/blob/master/api/lorawan.proto and https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/ttnpb/lorawan.go

#### Can you do this yourself and submit a Pull Request?

Yes, but need input from @rvolosatovs @johanstokking @htdvisser

rvolosatovs, 2020-03-30:
What input is required from us (@rvolosatovs @johanstokking @htdvisser)? This this issue missing a `discussion` label?

FTR, I think it depends on what fields we're talking about, but for LoRaWAN-specific MAC stuff I think we should use just the number, since that aligns with the spec (Ping slot periodicity of `PING_EVERY_2S` is not defined in the spec, however the one of `1` is)

htdvisser, 2020-03-30:
This is unfortunately not something that we can just change, as it would violate our API compatibility commitment, so this is definitely not something for March.

bafonins, 2020-03-31:
closed since we **wont fix** this

htdvisser, 2020-03-31:
I don't think we should immediately dismiss this as a wontfix. Let's keep it in the backlog and see if we can fix it in the future.

johanstokking, 2020-03-31:
Great one for V4