TheThingsNetwork/lorawan-stack #2269: `factory_preset_frequencies` behavior for OTAA

rvolosatovs, 2020-03-30:
#### Summary

`factory_preset_frequencies` behavior for OTAA is not obvious. It's only mentioned in the BI spec and it's stated that it's "mandatory for ABP", yet nothing is said about OTAA.

#### Why do we need this?

To get some clarity

#### What is already there? What do you see now?

NS assumes `factory_preset_frequencies` to be the channels device uses on reset, so if it's set to `A` and NS sends channels `B` in `CFList` of join-accept, NS assumes that device has channels `A ++ B` enabled immediately after join, which may or may not be expected by the user.

#### What is missing? What do you want to see?

a. Leave as-is, document
b. Disallow `factory_preset_frequencies` for OTAA
c. Ignore `factory_preset_frequencies` for OTAA
d. Assume `factory_preset_frequencies` are the join channels used by OTAA device

#### Can you do this yourself and submit a Pull Request?

yes

rvolosatovs, 2020-03-30:
Note, that we have very similar issues with RX1 data rate offset, RX2 data rate index and RX1 delay. (aka other MAC parameters being sent in join-accept)

rvolosatovs, 2020-03-30:
Note, that we have very similar issues with RX1 data rate offset, RX2 data rate index and RX1 delay

htdvisser, 2020-03-30:
> NS assumes that device has channels `A ++ B` enabled immediately after join

The effect of CFList is defined in PHY. For EU863-870 that is:

> _CFList is a list of five channel frequencies for the channels four to eight_

---

I like option (d) as it would allow devices to perform joins on those channels in addition to (or alternative to) the mandatory join channels.

The RX1 data rate offset and RX2 data rate are in fact defaults, so if those are specified, I think we should respect those (and use those for the join-accept).

The RX1 delay can be completely ignored for OTAA devices, because join-accept messages use JoinAcceptDelay1/2 instead of ReceiveDelay1/2.

rvolosatovs, 2020-03-30:
> I like option (d) as it would allow devices to perform joins on those channels in addition to (or alternative to) the mandatory join channels.

I think for consistency we should just assume device only has `factory_preset_frequencies` enabled in such case when joining regardless of the spec, using anything else except for the default join channels is against the spec anyway

> The RX1 data rate offset and RX2 data rate are in fact defaults, so if those are specified, I think we should respect those (and use those for the join-accept).

Sounds good

> The RX1 delay can be completely ignored for OTAA devices, because join-accept messages use JoinAcceptDelay1/2 instead of ReceiveDelay1/2.

The spec does have the following statement in RP
So IMO it makes perfect sense to allow the users to modify the join-accept delay.

[image]

johanstokking, 2020-03-31:
> So IMO it makes perfect sense to allow the users to modify the join-accept delay.

No we don't have to do this.

Otherwise, agreed with what's discussed above.

htdvisser, 2020-03-31:
> I think for consistency we should just assume device only has `factory_preset_frequencies` enabled

Agreed.

rvolosatovs, 2020-06-12:
Note, that we should also fix https://github.com/TheThingsNetwork/lorawan-stack/blob/01ee8f0155168faab49022a0acafb53116f8dd7d/pkg/networkserver/utils.go#L791 here and use "maximum possible DR idx" here instead

johanstokking, 2020-11-12:
Let's reschedule when we have a device that actually needs this.

rvolosatovs, 2021-03-10:
In current implementation, channels are simply replaced by the values of this field, but that breaks downlinks in US-like bands or, really, any band when uplink and downlink frequencies are not symmetric.
Rescheduling this and will fix along with OTAA

rvolosatovs, 2021-03-30:
If we don't implement this yet, I think a simple solution is to prohibit creation/update of this field for devices in US-like bands

htdvisser, 2021-03-30:
That sounds like a good plan, and I think it's even better than trying to support the field for these bands.

closed 2021-04-14 by adriansmares