TheThingsNetwork/lorawan-stack #2298: Type string byte in swagger file implies Base64 not Base16/Hex

jpmeijers, 2020-04-05:
# TL;DR
This issue is with Swagger, not with TTN. Swagger seems to only support Base64 for bytes, and no other formats like Base16. 

Perhaps one should define it as just strings in the Swagger file, and not specifically as bytes.

# Description
When one generate code according to the swagger file, the code will parse received DevAddr and DevEUI from TTN as Base64, while they are in fact Base16.

v3EndDeviceIdentifiers in the swagger file
https://github.com/TheThingsNetwork/lorawan-stack/blob/0478dd4d251c9ea5c7c4f2a1de0bb84093631bab/api/api.swagger.json#L9945-L9949

Compiles to the following go, which assumes Base64
```
	// The LoRaWAN DevAddr.
	// Format: byte
	DevAddr strfmt.Base64 `json:"dev_addr,omitempty"`
```

But the JSON I receive from the webhook looks like this, which is Base16/HEX
```
   "end_device_ids":{
      "device_id":"cricket-001",
      "application_ids":{
         "application_id":"jpm-crickets"
      },
      "dev_addr":"26011CE4"
   },
```

### Some references
https://github.com/go-swagger/go-swagger/issues/2170
https://github.com/OAI/OpenAPI-Specification/issues/50

### The proto definition where I assume the swagger definition is generated from
https://github.com/TheThingsNetwork/lorawan-stack/blob/0478dd4d251c9ea5c7c4f2a1de0bb84093631bab/api/identifiers.proto#L37-L48

Which refers to the DevAddr type, which uses Hex/Base16 to marshal to json
https://github.com/TheThingsNetwork/lorawan-stack/blob/0478dd4d251c9ea5c7c4f2a1de0bb84093631bab/pkg/types/devaddr.go#L60-L61

The same is true for the EUI type - it is also marshalled to HEX.

johanstokking, 2020-04-07:
Indeed, this is one of the early implementation choices we made, which breaks compatibility with JSONPB, i.e. what grpc-gateways interprets based on the protos from which the Swagger file is generated.

LoRaWAN specific types are indeed hex encoded in our API. We should somehow instruct the Swagger generator to reflect that.

---

PS please follow the issue template for future issues.

johanstokking, 2020-04-07:
References #32
References #1982
References #2258

jpmeijers, 2020-04-07:
> PS please follow the issue template for future issues.

When you browse the code on github, click on a line number, and then choose "Reference in new issue", the issue template is not filled in, so it's very difficult to keep to it using this workflow.

johanstokking, 2020-04-20:
@htdvisser can you give @neoaggelos some pointers to fix the Swagger file generation?

htdvisser, 2020-04-20:
That's all done by https://github.com/grpc-ecosystem/grpc-gateway

johanstokking, 2020-04-20:
Right, so are we forking this to adapt for proto fields indicating non-standard types?

neoaggelos, 2020-04-20:
Hey @jpmeijers 

May I ask what command are you using to generate code from the API swagger file?

johanstokking, 2020-04-20:
@neoaggelos you can try pasting our Swagger file [here](https://editor.swagger.io/) and use **Generate Client** in the top menu

neoaggelos, 2020-04-20:
This is what I am doing (also tried with the `swagger-codegen-cli` tool), and I get the `DevAddr` as a `string` field:

[image]

jpmeijers, 2020-04-20:
https://github.com/TheThingsNetwork/lorawan-stack/issues/2297#issuecomment-609378470

neoaggelos, 2020-04-20:
Okay, so this is an issue with https://github.com/go-swagger/go-swagger, as the original Java `swagger-codegen-cli` tool seems to properly generate the field with type `string`.

neoaggelos, 2020-07-13:
How should we proceed with this? cc @htdvisser @johanstokking

johanstokking, 2020-07-14:
Fork and fix, I'm afraid.

Alternatively, run a script to "fix" the generated Swagger files. That might be cleaner until we remove our custom sauce in V4.

jpmeijers, 2020-12-27:
I'm running into this same issue, as well as other related field type mismatch issues, when generating Kotlin client code from the Swagger file. I also tried using the proto files to generate Kotlin classes, but after a long struggle the generated code is unusable. Currently it seems like I'll have to make do with the code generated from the swagger file, and then manually go through them all and fix the types.

`./openapi-generator-cli.jar generate -g kotlin -i lorawan-stack/api/api.swagger.json`
generates models which defines the EUI fields as type ByteArray, but what I receive in json is HEX encoded strings.

johanstokking, 2020-12-28:
Yep, the generated Swagger is plain wrong. We should either pick this up or remove Swagger for the time being, because currently this is useless.

johanstokking, 2020-12-28:
References also https://github.com/TheThingsNetwork/lorawan-stack/issues/2798

jpmeijers, 2020-12-28:
Oh wow a mess this gogo situation is causing

jpmeijers, 2020-12-28:
As a side note, I find the swagger file still very helpful. The reason is that swagger can by compiled into simple POJO or other simple classes with which can then parse the TTI json objects - with some slight manual changes to them. Up until now I have not been able to build clean (no grpc/protocol buffers) classes for unmarshalling json from the proto files.

htdvisser, 2021-03-02:
Blocked on #2798. We're going to do some more refactoring to our protos and generated code.

closed 2022-07-20 by htdvisser