TheThingsNetwork/lorawan-stack #2312: Reverse AS-NS linking direction

adriansmares, 2020-04-06:
#### Background
_Can be skipped if you are familiar with https://github.com/TheThingsIndustries/lorawan-stack/issues/941_

In the current implementation of the communication between the Network Server and Application Server, the links are established by the Application Server, which links itself to the Network Server. Over this link, uplinks are sent as a stream of `ttnpb.ApplicationUp`.

The underlying issue with this approach is that Application Servers are inherently stateful due to linking. This introduces challenges with regards to load balancing between multiple Application Servers - the distribution of links over multiple instances, maintaining/migrating links when instances are down or a network partition occurs being just a number of issues introduced by the linking state.

#### Proposed solution

In order to tackle this issue, following our offline discussions, we've decided that we would like to reverse the direction of the links. This means that instead of the Application Server being a client to the Network Server, the Network Server becomes a client of the Application Server. This simplifies the logic for both components.

Since the Application Server may have clients (MQTT/gRPC) connect to it while not being linked by a Network Server, a new PubSub service should be used as a broker between Application Server instances. When a new application subscription (connection) arrives to an Application Server, a PubSub subscription is created for the uplinks of the said application. When an uplink is received by an Application Server instance, it is published to the topic of the application.

Application Server PubSub integrations should be launched/connected to the external PubSub service only while having an active link from a Network Server instance. This should ensure that uplinks and downlinks are no duplicated by multiple subscriptions/publishings. 

#### Implementation details:

- `LinkApplication` should be added to a new `NsAs` service that the Application Server implements.
```proto
service NsAs {
  rpc LinkApplication(stream ApplicationUp) returns (stream google.protobuf.Empty);
}
```
> We may want to revisit https://github.com/TheThingsNetwork/lorawan-stack/issues/1523 since we are already adding a new RPC.
- The NS uses this RPC in order to push uplinks to the AS.
- The NS uses `component.GetPeer`/`component.GetPeerCon` in order to receive the peer/connection to the AS based on the application identifiers.
- When the AS receives an uplink, it triggers the required webhooks and application packages. Only the AS that receives the uplink should send the webhooks and run the application packages. 
The uplink is then published to the PubSub service, from where subscription based frontends can pick it up.
- AS PubSub integrations are now bound to an NS connection. This means that the AS connects to the external PubSub integration server only when the NS starts a `LinkApplication` stream, and the integration has the same lifetime as the stream.
- AS downlink queue operations no longer use the link connection, but rather the cluster peer connection.
- AS can use `gocloud.dev/pubsub` for PubSub communication for inter-AS PubSub.

#### Required migrations
We need to migrate the application default payload formatters to a more general Application Server application store. The migration should be lazy - when we need the application payload formatter we check if we have it in the AS application store, and if it is missing we attempt to retrieve it from the old link store. Any writes/updates are done in the application store.

#### Open questions
- Should we want to drop external NS linking completely ?
- Which PubSub service should we bundle by default ? I personally propose NATS, but I do not have strong feelings about this.

rvolosatovs, 2020-04-06:
Why not `NsAs.HandleUplink` as we discussed before? (https://github.com/rvolosatovs/lorawan-stack/commit/a514740e9ac07fd93b9b2741a5642b8af10f95c7#diff-23d97d1622d559c2eede816e5a8af414R105-R109). (FYI: I changed the type to accept a list of `ApplicationUp` in a more recent revision)

We want to be able to load-balance these and to be able to do things concurrently. Stream mapping a NS to 1 singular AS is definitely not a way to go.

> Should we want to drop external NS linking completely ?

Yes, that's what we agreed upon

adriansmares, 2020-04-06:
> 
> 
> Why not `NsAs.HandleUplink` as we discussed before? ([rvolosatovs@a514740#diff-23d97d1622d559c2eede816e5a8af414R105-R109](https://github.com/rvolosatovs/lorawan-stack/commit/a514740e9ac07fd93b9b2741a5642b8af10f95c7#diff-23d97d1622d559c2eede816e5a8af414R105-R109)). (FYI: I changed the type to accept a list of `ApplicationUp` in a more recent revision)
> 
> We want to be able to load-balance these and to be able to do things concurrently. Stream mapping a NS to 1 singular AS is definitely not a way to go.
> 
> > Should we want to drop external NS linking completely ?
> 
> Yes, that's what we agreed upon. The goal is to get rid of linking altogether.

I've updated the issue. _Should've written this earlier in order to keep all the details in mind_.

rvolosatovs, 2020-04-07:
> We may want to revisit #1523 since we are already adding a new RPC.

I already started on this locally couple of days ago including NS support. Let's do this together with this issue, otherwise later we'll need to deprecate yet another RPC, which is not worth it in this case, since it's such a trivial feature to add with `NsAs` interface in place.

johanstokking, 2020-04-07:
Great.

> * The NS uses `component.GetPeer`/`component.GetPeerCon` in order to receive the peer/connection to the AS based on the application identifiers.

We wouldn't tie the AS instance to application identifiers, right? We should be spreading traffic from applications with many devices over multiple AS instances.

> * AS can use `gocloud.dev/pubsub` for PubSub communication for inter-AS PubSub.

I like the idea of AS-AS communication via pub/sub. For the future, I think we should document why we chose AS-AS instead of a dedicated component for applications to subscribe to.

> * Which PubSub service should we bundle by default ? I personally propose NATS, but I do not have strong feelings about this.

We can use the in-memory variant by default for single process deployments and the getting started, see `mempubsub`. Indeed, for multiple processes, something lightweight makes most sense, and NATS seems the best candidate there.

> * How can AS PubSub integrations handle downlink operations ? Internally we can ensure that uplinks are sent only once, using subscription groups, but I don't see how we can deduplicate downlink queue operations, since some services (MQTT) do not support subscription groups.

What scenario is this? That an application schedules downlink duplicates via multiple ways, i.e. MQTT and webhooks? I don't think we should/want to do something against that.

adriansmares, 2020-04-07:
> We wouldn't tie the AS instance to application identifiers, right? We should be spreading traffic from applications with many devices over multiple AS instances.

True - probably `based on the application identifiers` can be left out here.
 
> What scenario is this? That an application schedules downlink duplicates via multiple ways, i.e. MQTT and webhooks? I don't think we should/want to do something against that.

No, _and now I see that the original text wasn't clear_. It's about which instance starts which PubSub integrations (that we currently have with MQTT and NATS) in multi AS environments. We cannot have multiple instances subscribe to the same external PubSub service, since then downlink queue operations would be duplicated:
- Two AS instances
- AS1 connects to NATS1 
- AS2 connects to NATS1 (same integration if we don't don't some kind of distribution)
- User publishes to NATS a downlink push
- Both AS pick it up and push the downlin twice

htdvisser, 2020-04-14:
What is still up for discussion here? Can we remove the label?

johanstokking, 2020-04-14:
I see what you mean. Yes, that is something to account for. We need some sort of sharding. That touches however on clustering, which we don't discuss here.

What else needs to be discussed here to move forward with this?

adriansmares, 2020-04-14:
For now all of the open questions have been addressed (to the extent of this repository). I will remove the `discussion` label.

adriansmares, 2020-09-04:
I've pushed my current progress in https://github.com/TheThingsNetwork/lorawan-stack/pull/3190 . I think I covered most (if not all) of the AS implementation, but I have some questions regarding the API:

- Should a `HandleUp` request contain upstream messages from any application, or just one application ? @johanstokking @rvolosatovs . FTR I don't see any dangers in having non-homogeneous requests, but I would like to clarify this aspect.
- Can we include the name of the NS in the request ? It would be useful for telemetry. @rvolosatovs 
- What do we do with the `GetLinkStats` call ? Does it return an error ? Does it return an empty response ? @johanstokking

johanstokking, 2020-09-08:
> * What do we do with the `GetLinkStats` call ? Does it return an error ? Does it return an empty response ? @johanstokking

- Return `Unimplemented`
- Remove from CLI
- Remove usage from Console

PR #3263 (2020-11-27): Remove AS-NS linking in the AS and NS

#### Summary

Closes #2312 
References #3190 
References #3260 
Closes https://github.com/TheThingsNetwork/lorawan-stack/issues/3145

#### Changes

- Reintroduce the changes added by #3190

#### Testing

Covered by #3190 

##### Regressions

Covered by #3190 

#### Notes for Reviewers

I merged #3190 too fast, which left the `v3.10` branch in an unusable state. As such, I've reverted the merge in #3260 in order to make the branch usable again. 

This PR was generated using `git revert -m 1 62f404a75c37a93b431b0133927bcfd773404c4f` on 62f404a75.

As it was [... cut]
closed 2020-11-27 by adriansmares