TheThingsNetwork/lorawan-stack #2316: Frontend errors are inconsistent

pgalic96, 2020-04-07:
#### Summary

Sometimes there are errors that contain only `code` attribute, which in turn makes the `isBackend` (https://github.com/TheThingsNetwork/lorawan-stack/blob/fc193ed01286a4638d938effc0e882cdc9071d19/pkg/webui/lib/errors/utils.js#L23) false because it is not translated. However, we should still be able to retrieve `httpStatusCode` (https://github.com/TheThingsNetwork/lorawan-stack/blob/fc193ed01286a4638d938effc0e882cdc9071d19/pkg/webui/lib/errors/utils.js#L55) since the `code` attribute indicates grpc code which can be mapped to the http equivalent. That is currently not possible because in case `isBackend` is false , only `statusCode` is checked. Another issue is that `httpStatusCode` can return both string and number based on the way it is retrieved.
#### Steps to Reproduce

1. Log `httpStatusCode(e)` before https://github.com/TheThingsNetwork/lorawan-stack/blob/f32761880d11daa7188fdda761e7bfe90f82cece/pkg/webui/console/store/middleware/logics/lib.js#L85
2. Go to gateway general settings
3. Input a very large number for Schedule Any Time Delay (this is another issue) and look at the console.

#### What do you see now?

The error returned is 400, with the error returned with grpc code 3, which is invalid argument error. This mapping already exists in `grpcToHttpErr` so we should still be able to retrieve the error. I guess this partly concerns also the way backend formats this error message, without details, so I think this should be fixed in backend first cc @htdvisser 

This issue is important to fix in order to correctly filter which messages to report to sentry frontend (#2264)

#### What do you want to see instead?

I want `httpStatusCode` to correctly return the 400, in a number format always.

#### Environment

Current master

#### How do you propose to implement this?

Either fix the logic for recognizing these errors in frontend, or make sure that error codes returned from the backend are consistent and have all the required fields.

#### Can you do this yourself and submit a Pull Request?

Yes, but we can either handle these inconsistencies in frontend or the error codes should be fixed in backend first. Need some advice on this cc @bafonins @kschiffer

kschiffer, 2020-04-08:
I'm fine with adapting our error utils as a quick fix, but this should indeed also be addressed in the backend.

htdvisser, 2020-04-08:
I'm not sure we'll be able to define and translate all possible errors in the backend, so the frontend definitely needs to handle errors gracefully, starting with a basic error, and enriching them when more details are available. 

- All errors have an HTTP status code, so that should be the base error. This type of error can occur when there's an issue with the load balancer, proxy or maybe the backend is under maintenance. In this case we can only rely on the status code, the body may contain a (text or json) error.
- Most errors are JSON objects with a `code` and `message`. The `code` is the gRPC status code which should typically correspond to the HTTP status code. The `message` is something that can be displayed to the user.
- "TTN errors" have an object in `details` with `"@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails"` (remember that this may not always be the first element of `details`). When errors contain details they can be translated and formatted.

---

If you run into common errors that you think need to be translated, please create issues.

closed 2021-12-17 by NicolasMrad