TheThingsNetwork/lorawan-stack #2323: Inconsistent responses on end device deletion requests

bafonins, 2020-04-08:
#### Summary

IS and JS behave differently from NS and AS on repeated end device deletion.

#### Steps to Reproduce

1. Create a new end device
```
// delete existing entry from is:
$ curl -X DELETE -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/applications/test-app/devices/test-dev2 | jq

{}

// try again:
$ curl -X DELETE -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/applications/test-app/devices/test-dev2 | jq

{
  "code": 5,
  "message": "error:pkg/identityserver/store:end_device_not_found (end device `test-app:test-dev2` not found)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/identityserver/store",
      "name": "end_device_not_found",
      "message_format": "end device `{application_id}:{device_id}` not found",
      "attributes": {
        "application_id": "test-app",
        "device_id": "test-dev2"
      },
      "correlation_id": "1899b283a1f847e0b5ddd328fdf9f713",
      "code": 5
    }
  ]
}

// delete existing entry from js:
$ curl -X DELETE -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/js/applications/test-app/devices/test-dev2 | jq

{}

// try again:
$ curl -X DELETE -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/js/applications/test-app/devices/test-dev2 | jq

{
  "code": 5,
  "message": "error:pkg/joinserver:device_not_found (device not found)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/joinserver",
      "name": "device_not_found",
      "message_format": "device not found",
      "correlation_id": "73d3a76c6b04481395ac0aba5a04531c",
      "code": 5
    }
  ]
}

// delete existing entry from ns:
$ curl -X DELETE -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/ns/applications/test-app/devices/test-dev2 | jq

{}

// try again:
$ curl -X DELETE -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/ns/applications/test-app/devices/test-dev2 | jq

{}

// delete existing entry from as:
$ curl -X DELETE -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/as/applications/test-app/devices/test-dev2 | jq

{}

// try again:
$ curl -X DELETE -H "Authorization: Bearer $access_token" http://localhost:1885/api/v3/as/applications/test-app/devices/test-dev2 | jq

{}
```

#### What do you see now?

repeated deletion requests to IS and JS return 404 and an error, while the same requests to NS and AS return 200

#### What do you want to see instead?

Consistent responses between stack components

#### Environment

v3.7.0

#### How do you propose to implement this?

@htdvisser @rvolosatovs @johanstokking 

#### Can you do this yourself and submit a Pull Request?

Yes if this is needed

htdvisser, 2020-04-08:
As far as I know our entire API returns errors based on whether the action succeeded, and not on what the state after the action is. This means that:

- You can not `get` an X that does not exist
- You can not `create` an X that already exists
- You can not `update` an X that does not exist
- You can not `delete` an X that does not exist

In our API we also have a `set`, which is allowed both when an X does and doesn't exist.

From this, it follows that the NS and AS currently return incorrect responses for `delete`.

neoaggelos, 2020-04-11:
Picking this one up, if that is OK.

neoaggelos, 2020-04-11:
Also related to #259

johanstokking, 2020-04-13:
Keeping @rvolosatovs assigned to make sure the device registries are behaving consistently.

bafonins, 2020-04-20:
Closing since https://github.com/TheThingsNetwork/lorawan-stack/pull/2342 and https://github.com/TheThingsNetwork/lorawan-stack/pull/2343 are merged

closed 2020-04-20 by bafonins